MarlinFirmware/Configurations #2: Status/plans of this repository

bojanpotocnik, 2019-11-17:
When I first saw this repository I thought that the `MarlinFirmware/Marlin/config/examples` directory will be removed at the expense of this repository used for that purpose; or that `Marlin/config` will  be a git-submodule of `MarlinFirmware/Configurations` and then each user will checkout the desired submodule branch; or .... .

It seems, however, that this repository is not actively maintained, neither are the configuration files correct (looking in some directories all the Configuration.h are just copies of the default).

What are the plans if this repo? I suppose we shall use the files in `MarlinFirmware/Marlin/config/examples` until further notice?

Dakkaron, 2019-11-22:
I reckon you are on 1.9.x? On the 2.0.x the example configuration folder has been removed already.

RudyBenner, 2019-11-22:
It has been moved, not removed.
C:\Users\RudolfBenner\Documents\GitHub\Marlin\config

Dakkaron, 2019-11-22:
Thanks for correcting that, I didn't see that.

RudyBenner, 2019-11-22:
Can you close?

Dakkaron, 2019-11-23:
The original question wasn't answered.

bojanpotocnik, 2019-11-23:
As RudyBenner already noted, I am talking about `bugfix-2.0.x` where the `Marlin/config/examples/` directory is updated almost daily, in contrast to this repo which is not updated at all (the only 4 commits are the Initial commit, Import of configs and two README updates).

This idea of having the separate repository is good, but I am also aware that it is harder to realize it - to migrate the existing structure to this new layout. That's why I am wondering, what are the plans? Is this some afternoon idea with the lowest priority among Marlin contributors, is someone already actively working on, is there any help required, ...?

thisiskeithb, 2020-01-15:
Check out https://github.com/MarlinFirmware/Marlin/commit/7b304cee16bf87962eba998d8c293e5d4bd23c71. Configs have been moved here 🙂

closed 2020-01-15 by bojanpotocnik