MarlinFirmware/Configurations #8: ANET A8Plus configuration

devond76180, 2020-01-20:
This configuration will make a mess of the screen on the A8plus.  If you look at the A6 configuration line 1997 and E10 configuration line 1881 they have setup ST7920_DELAY_?.  This should also exist in the A8plus configuration.  A quick fix is to set it to the CR10_STOCKDISPLAY

thinkyhead, 2020-01-23:
Is `CR10_STOCKDISPLAY` required? Or is it better to use…
```cpp
#define ANET_FULL_GRAPHICS_LCD
#define ST7920_DELAY_1 DELAY_NS(150)
#define ST7920_DELAY_2 DELAY_NS(150)
#define ST7920_DELAY_3 DELAY_NS(150)
```
…?

veeectorm, 2020-05-10:
> This configuration will make a mess of the screen on the A8plus. If you look at the A6 configuration line 1997 and E10 configuration line 1881 they have setup ST7920_DELAY_?. This should also exist in the A8plus configuration. A quick fix is to set it to the CR10_STOCKDISPLAY

Im running 2.0.5.3 on my Anet A8 Plus on Anet board v1.7. No issues at all with the display.

thinkyhead, 2020-05-12:
Different setups may have more noise around the LCD wires. Increase delays as needed.

closed 2020-05-12 by thinkyhead