MarlinFirmware/Configurations #13: CR10 V2 Configuration

zigomatichub, 2020-01-24:
Dears,
I'm searching the configuration example of my printer which is a pretty new model: CR10 V2.
What I need to do in order to help you to have the work done ?

thinkyhead, 2020-02-15:
If you find one out there in your searches, please let us know. We don't yet have this on the radar. Maybe ask TinyMachines3D about it.

NathanCrainJr, 2020-03-12:
I have done some digging around and found these files in Creality's dropbox linked here [https://www.dropbox.com/sh/dt38tetxsxowg7b/AAD_kCsmkdlgafImo9e_gKKia?dl=0](url)
It is clearly an old version of Marlin with the CR-10 V2 applicable settings. Would this help anyone who could update the configuration.h and configuration_adv.h to the current Marlin version?

APajonk, 2020-03-12:
I´m also currently searching for a way to update my new CR-10 v2 to current Marlin firmware. Did not found any existing HowTo or any description about this.

NathanCrainJr, 2020-03-14:
Update: I have recently found this Cheetah 5.0 firmware that have specific settings for CR-10 V2. I am going to see which parameters they have used for the machine and try to use those in vanilla Marlin.
https://kay3d.com/pages/kay3d-cheetah-5-firmware
There is a facebook group specifically for the CR-10 V2, but they only share a .Hex file. I do not know how to reverse engineer this file to extract settings. It runs perfectly on my machine though.

Chrissko, 2020-03-20:
I have a similar problem with the CR-10 Pro V2. I think that the config should be very close. So I would be very interested in progress here. ;)

zigomatichub, 2020-03-22:
I guy on a forum which have a CR10V2 gave me his Configuration.h , Configuration_adv.h, _Bootscreen.h and _Statusscreen.h. All modifications done for Marlin V2 and CR10V2.
I don't have look right now but his build is working fine on his printer. I don't know how I can share those files here

NathanCrainJr, 2020-03-22:
You linked the tinymachines3d firmware. It works pretty darn good. Unfortunately, the config and config_adv files are special tailored to the way they modified the marlin firmware.

swilkens, 2020-04-20:
This website:
https://www.creality3dofficial.com/pages/firmware-download

Has `CR-10 V2 Firmware + BL Touch` - that link also includes an archive with the sources, including the configuration files.

thinkyhead, 2020-04-23:
I will have to compare it to the sources provided to me. I am in the process of bringing it up to date with bugfix-2.0.x so that all the kinks are worked out. The starting date of their sources is from a very brief period where the `platformio.ini` file was using inheritance to be hierarchical.

mdmartini, 2020-04-23:
I also just purchased a CR10 V2, and would like to update to the lastest version of vanilla Marlin to get the new features (mostly the thermal runaway), but I seem to be having issues with errors, and I think it is because I have to either select 'CR10S' or CR10.     I have gotten a CR10S config to upload once, however it now shows 'CR10S' on the info screen, and i am not completely sure it is the correct set up.     Is it possible that a new version of Marlin will have a config specifically for the CR10 V2?

zigomatichub, 2020-04-24:
There is a french guy that I done a marlin for cr10 V2 based on 2.0.5.3 Marlin.
Here the reference with all details to redo the same.

https://www.lesimprimantes3d.fr/forum/topic/29980-cr10-v2-carte-originelle-252-marlin-2053/?tab=comments#comment-348341

thinkyhead, 2020-04-25:
I also have the Creality3D source in hand, partially ported to the latest code. I'll take a look at the french guy version.

thinkyhead, 2020-04-28:
I've ported Ender 3 V2 to the latest `bugfix-2.0.x` but it's not tested. #17719

I have to figure out how to flash it onto the machine. The provided method is "jlink" but I don't have one of those devices. I tried "stlink" with my ST-Link device, but that just gave the same errors as jlink. Putting the `firmware.bin` file on the card doesn't work. And USB/serial doesn't work either. I'll have an answer from Creality soon.

kiss81, 2020-06-14:
Any update on this? It would be nice to have this work on a CR-10 v2.

thinkyhead, 2020-06-15:
Just added this… 9209d64dc86be02ed81d5ac1baaa1642d8479d3e

kiss81, 2020-06-15:
> Just added this… [9209d64](https://github.com/MarlinFirmware/Configurations/commit/9209d64dc86be02ed81d5ac1baaa1642d8479d3e)

Great! I will give it a go.

closed 2020-06-15 by thinkyhead

kiss81, 2020-06-16:
Update: the build succeeded, but after uploading I only see the blue LCD lights turning on and nothing seems to be running unfortunately. The bootloader was still ok, so I could upload the stock firmware and everything was fine. Anyone got a clue?

thinkyhead, 2020-06-17:
You wouldn't be able to flash any firmware if the board didn't have a bootloader.

This configuration is indeed untested, and has simply been ported over from fran6p. I suggest you ask him to try the latest Marlin firmware with these configs, and he'll probably figure out what is missing or incorrect pretty quickly, at least to the point of starting up.

I only converted the configuration files. I didn't see any other firmware changes described in his postings.

kiss81, 2020-06-17:
I compared the new configuration with the original stock version, but couldn't find anything that should block it from booting. I tested some changes without luck. I will contact fran6p

kiss81, 2020-06-19:
@fran6p
@thinkyhead

I finalized a version that works on the CR-10 v2 stock setup:
https://github.com/kiss81/cr10v2
in the "marlin" folder.

The main issue I struggled with where some RAM issues and inverted END stops in the Z direction... That caused several issues of course 🙂
I am trying to do some prints now to see how it goes.

thinkyhead, 2020-06-20:
I'll download yours and compare with https://www.lesimprimantes3d.fr/forum/messenger/8107/?tab=comments#comment-67039 and merge the final result soon.

kiss81, 2020-06-20:
> I'll download yours and compare with https://www.lesimprimantes3d.fr/forum/messenger/8107/?tab=comments#comment-67039 and merge the final result soon.

The latest version I just uploaded is working well now. There are some changes which are personal preference, but the z clearances as they are now are at least really safe to work with.

thinkyhead, 2020-06-21:
Some minor differences between the two versions…
May I assume the newer one is correct in all cases?

(For example, the newer one has all Z endstop inverting set to `false` and the older one has them set to `true`.)

fran6p, 2020-06-21:
Hi Scott,
Without ABL (Bltouch), the logic is «false» for :

Z_MIN_ENDSTOP_INVERTING
Z_MIN_PROBE_ENDSTOP_INVERTING

With ABL the two are «true»

:-)

kiss81, 2020-06-21:
Exactly. Make sure not to use interrupts for the end stops as that will not always work...

thinkyhead, 2020-06-21:
Please review the currently published version for any needed corrections.

kiss81, 2020-06-22:
I don't have bltouch, so can't test it, but the changes seems legit for a working version. The default extruder steps are wrong though. They are wrong in every firmware I've seen so far. It should be ~97. (after calibration my unit is even at 98.8, but it depends a bit per unit)

#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 400, 97 }

fran6p, 2020-06-23:
I have reviewed the files.
All seems Ok for me BUT my configuration is for a CR10-V2 with BLTouch (and
not the standard extruder). I have tested it and it works for me and some
other french users of this forum (lesimprimantes3d.fr).
@kiss81 you're right with the steps for the extruder, it should be 95 and
the user should do a calibration to find the steps for his own printer.
On my CR10-V2, before I added BLTouch and change my extruder with the same
one found on CR10S Pro (a pseudo BMG without reduction whose steps are near
140) after calibration my extruder's steps where between 94-95 (I don't
remember exactly… Maybe I'm becoming too old ;-) .
@scott I have to thank you for all the work you (and others) do for the
Marlin software.
:-)

DavBfr, 2020-06-28:
@fran6p Is it possible to add some conditions to set the right settings depending on the presence of the BLTOUCH?
like:

```C
#if ENABLED(BLTOUCH)
  #define Z_MIN_ENDSTOP_INVERTING true
  #define Z_MAX_ENDSTOP_INVERTING true
  #define Z_MIN_PROBE_ENDSTOP_INVERTING true
#else
  #define Z_MIN_ENDSTOP_INVERTING false
  #define Z_MAX_ENDSTOP_INVERTING false
  #define Z_MIN_PROBE_ENDSTOP_INVERTING false
#endif
```

This way it's easier to build the correct version.

kiss81, 2020-06-28:
> @fran6p Is it possible to add some conditions to set the right settings depending on the presence of the BLTOUCH?
> like:
>
> ```c
> #if ENABLED(BLTOUCH)
>   #define Z_MIN_ENDSTOP_INVERTING true
>   #define [...]

I agree, such improvements makes things so much easier. If needed I can make a commit on this that you can compare.

fran6p, 2020-06-29:
Hi,
Good suggestion @david PHAM-VAN
I have tested it with :

>
> // Mechanical endstop with COM to ground and NC to Signal uses "false" here (most common setup).
> #define X_MIN_ENDSTOP_INVERTING false
>  // Set to true to invert the logic of the endstop.
> #define Y_MIN_ENDSTOP_INVERTING false
>  // Set to true to invert the logic of the endstop.
>
> //#define Z_MIN_ENDSTOP_INVERTING true // Set to true to invert the logic of the endstop. //Test
> #define X_MAX_ENDSTOP_INVERTING false
>  // Set to true to invert the logic of the endstop.
> #define Y_MAX_ENDSTOP_INVERTING false
>  // Set to true to invert the logic of the endstop.
>
> //#define Z_MAX_ENDSTOP_INVERTING true // Set to true to invert the logic of the endstop. //Test
>
> //#define Z_MIN_PROBE_ENDSTOP_INVERTING true // Set to true to invert the logic of the probe. //Test
>
> // Test suggestion David PHAM-VAN
> #if ENABLED(BLTOUCH)
>   #define Z_MIN_ENDSTOP_INVERTING true
>   #define Z_MAX_ENDSTOP_INVERTING true
>   #define Z_MIN_PROBE_ENDSTOP_INVERTING true
> #else
>   #define Z_MIN_ENDSTOP_INVERTING false
>   #define Z_MAX_ENDSTOP_INVERTING false
>   #define Z_MIN_PROBE_ENDSTOP_INVERTING false
> #endif
>
The compilation with VSC is OK even though these lines (near lines 654) are
before the #define BLTOUCH (line 987 on my configuration.h), I'm not a
coding professional ;-)
So if you want @kiss81 you can make a commit as I don't know how to do it
and have no time now to learn as I just began to use Github (now on my
CR10-V2 I test / use Klipper but I'm on Marlin with my CR10 (old one)).

Cordially
Francis Poulizac

alittlec, 2020-10-18:
Hi,
This doesn't build with the latest 2.0.7.2 release:

buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:1054:19: error: #if with no expression

wirecuttr, 2020-12-25:
Hi there, Looking at the lastest Configuration.h in the bugfix branch. I'm seeing:
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

Did you guys bypass the z-endstop switch and wire the BLTOUCH to the Z_MIN?