MarlinFirmware/Configurations #14: MKS Base 1.6 and TFT35 configuration

svenddk, 2020-01-26:
Marlin 2.0.x configuration files for the following 3D printer setup:

- Controller: Makerbase MKS Base v1.6, to add more I/O and memory then original Anet A6 controller board
- Display: Original Anet 12864 display or replacement by Makerbase MKS-TFT35 host controller, wifi card is optional. Both are in the configuration files, but the files are default for MKS-TFT35 - if using 12864 then you need to modify the enclosed files.
- Z-endstops: Using dual (left and right side) endstops to ensure level X-axis, no need for manual alignment of Z motor heights
- Frame: Original Anet A6 acrylic frame or a customized 2040 profiles aluminum frame, which is more robust and rigid
- Heatbed: Original Anet A6 factory size heatbed or a customized 250x250x5 mm heat-treated aluminum plate, which is absolutely flat

Warning: Standard wiring of Anet 12864 display does not fit to MKS Base 1.6 by default because Anet switched 5V and ground. You therefore need to switch 2 wires in the flatcable before connection and power-up. Ignoring this will damage the Anet 12864 display! The wiring is explained in the files, and guides are widely available on the web - just search for it.

In all the files my mods are marked with "svenddk" to make it easier for you to find the tweaks.

All files are provided as is, confirmed fully functional on my Anet A6 MKS Base v1.6 MKS-TFT35 2040 frame tweaked 3D printer. Use files at your own risk.

Files are zipped here:
[2.0.x.zip](https://github.com/MarlinFirmware/Configurations/files/4113248/2.0.x.zip)

thinkyhead, 2020-02-15:
Thank you very much. I didn't expect submissions so soon. We'll check them over and incorporate them ASAP.

svenddk, 2020-02-15:
You're welcome, thank you for the great efforts by the Marlin team!

closed 2021-11-10 by svenddk