MarlinFirmware/Configurations #15: Releases to correlate with Marlin Stable releases

XBrav, 2020-02-03:
With Marlin 2.0.3, the configuration examples directory points to this git. However, as config files are being updated here as per the bugfix, there are already compilation conflicts between 2.0.3 and 2.0.x bugfix. For example, LEVEL_CORNERS_INSET has been modified to the array LEVEL_CORNERS_INSET_LFRB.

If the configurations are no longer being distributed under the main package, it would be extremely valuable to release a stable example package in line with Marlin releases.

thinkyhead, 2020-02-04:
Thanks for the note! I've re-jiggered my scripts and regenerated the repo so that `master` will remain the default branch and will always contain the configs for the latest release. For CI on the most recent unreleased updates the branch `bugfix-2.0.x` will now be used.

stelgenhof, 2020-02-15:
Experienced the same issue with my MKS SBASE (LEVEL_CORNERS_INSET missing warning). For the moment, I've added LEVEL_CORNERS_INSET back in just to make the compilation happening :)

swilkens, 2020-02-25:
Current marlin release 2.0.4.3 which has a readme file that links here currently does not build with 2.0.4 release of this repo.

The repo readme states 2.0.3 is the latest, we only see a 2.0.4 exists if we look at the tags.

I understand there are some growing pains, switching to a new setup for configs - but I wonder if this is more complicated than it needs to be.

closed 2020-03-14 by thinkyhead