MarlinFirmware/Configurations #17: 15 example from Bugfix 2.0.x Configuration files fail to compile with marlin bugfix 2.0.x

ellensp, 2020-02-12:
New git clones of Marlin bugfix 2.0.x and Configurations where taken about 2 hours ago.

A script of my own was run that goes through all configurations and using marlins auto_build.py
attempt to compile them.

15 configuration files fail to build with an assortment of errors

Trying:: Configurations/config/examples/AlephObjects/TAZ4/
==================== 1 failed, 0 succeeded in 00:00:02.586 ====================
Trying:: Configurations/config/examples/Micromake/C1/enhanced/
==================== 1 failed, 0 succeeded in 00:00:02.328 ====================
Trying:: Configurations/config/examples/3DFabXYZ/Migbot/
==================== 1 failed, 0 succeeded in 00:00:02.297 ====================
Trying:: Configurations/config/examples/Formbot/Raptor/
==================== 1 failed, 0 succeeded in 00:00:30.194 ====================
Trying:: Configurations/config/examples/VORONDesign/
==================== 1 failed, 0 succeeded in 00:00:02.509 ====================
Trying:: Configurations/config/examples/STM32/STM32F4/
==================== 1 failed, 0 succeeded in 00:00:15.007 ====================
Trying:: Configurations/config/examples/STM32/Black_STM32F407VET6/
==================== 1 failed, 0 succeeded in 00:01:43.783 ====================
Trying:: Configurations/config/examples/Tronxy/X5S-2E/
==================== 1 failed, 0 succeeded in 00:00:02.592 ====================
Trying:: Configurations/config/examples/makibox/
==================== 1 failed, 0 succeeded in 00:00:03.925 ====================
Trying:: Configurations/config/examples/FYSETC/F6_13/
==================== 1 failed, 0 succeeded in 00:00:02.451 ====================
Trying:: Configurations/config/examples/FYSETC/S6/
==================== 1 failed, 0 succeeded in 00:01:41.950 ====================
Trying:: Configurations/config/examples/Malyan/M200/
==================== 1 failed, 0 succeeded in 00:00:02.502 ====================
Trying:: Configurations/config/examples/TheBorg/
==================== 1 failed, 0 succeeded in 00:00:14.125 ====================
Trying:: Configurations/config/examples/tvrrug/Round2/
==================== 1 failed, 0 succeeded in 00:00:02.466 ====================
Trying:: Configurations/config/examples/ArmEd/
==================== 1 failed, 0 succeeded in 00:01:41.240 ====================

full log of errors.
[marlin.log](https://github.com/MarlinFirmware/Configurations/files/4191885/marlin.log)

thinkyhead, 2020-02-14:
I'll do my best to get in touch with the authors of each of them for validation while waiting for patches.

thinkyhead, 2020-02-15:
Seems all the issues were minor, so no need to bug those maintainers. STM32/STM32F4 and TheBorg might only be buildable in Arduino IDE. But possibly that could be addressed in the future. The STM32/Black_STM32F407VET6, FYSETC/S6, and ArmEd configs didn't exhibit any problems in my build tests.

Thanks for narrowing it down. At some point this will get automated. For now I just use scripts…
```sh
use_example_configs ArmEd ; mftest -b
```

closed 2020-02-15 by thinkyhead