MarlinFirmware/Configurations #21: Malayan M200 fails when linking.

freiform, 2020-02-20:
I get the follwing error when trying to build for the Malayan 200, following https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/config/examples/Malyan/M200 to the letter. This happens with bugfix-2.0.x as well as with 2.0.x.

```
Linking everything together...
"C:\\Users\\s\\AppData\\Local\\Arduino15\\packages\\STM32\\tools\\xpack-arm-none-eabi-gcc\\9.2.1-1.1/bin/arm-none-eabi-gcc" -mcpu=cortex-m0 -mthumb -Os --specs=nano.specs -Wl,--defsym=LD_FLASH_OFFSET=0x2000 -Wl,--defsym=LD_MAX_SIZE=122880 -Wl,--defsym=LD_MAX_DATA_SIZE=15168 -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--entry=Reset_Handler -Wl,--unresolved-symbols=report-all -Wl,--warn-common "-TC:\\Users\\s\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.8.0\\variants\\MALYANMx00_F070CB/ldscript.ld" "-Wl,-Map,e:\\tmp/Marlin.ino.map" "-LC:\\Users\\s\\AppData\\Local\\Arduino15\\packages\\STM32\\tools\\CMSIS\\5.5.1/CMSIS/DSP/Lib/GCC/" -larm_cortexM0l_math -o "e:\\tmp/Marlin.ino.elf" "-Le:\\tmp" -Wl,--start-group "sketch\\Marlin.ino.cpp.o" "sketch\\SrcWrapper.cpp.o" "sketch\\src\\HAL\\HAL_STM32\\HAL.cpp.o" "sketch\\src\\HAL\\HAL_STM32\\HAL_SPI.cpp.o" "sketch\\src\\HAL\\HAL_STM32\\Sd2Card_sdio_stm32duino.cpp.o" "sketch\\src\\HAL\\HAL_STM32\\Servo.cpp.o" "sketch\\src\\HAL\\HAL_STM32\\SoftwareSerial.cpp.o" "sketch\\src\\HAL\\HAL_STM32\\fastio.cpp.o" "sketch\\src\\HAL\\HAL_STM32\\persistent_store_flash.cpp.o" "sketch\\src\\HAL\\HAL_STM32\\persistent_store_impl.cpp.o" "sketch\\src\\HAL\\HAL_STM32\\persistent_store_sdcard.cpp.o" "sketch\\src\\HAL\\HAL_STM32\\timers.cpp.o" "sketch\\src\\HAL\\HAL_STM32\\watchdog.cpp.o" "sketch\\src\\HAL\\shared\\HAL_spi_L6470.cpp.o" "sketch\\src\\HAL\\shared\\backtrace\\backtrace.cpp.o" "sketch\\src\\HAL\\shared\\backtrace\\unwarm.cpp.o" "sketch\\src\\HAL\\shared\\backtrace\\unwarm_arm.cpp.o" "sketch\\src\\HAL\\shared\\backtrace\\unwarm_thumb.cpp.o" "sketch\\src\\HAL\\shared\\backtrace\\unwarmbytab.cpp.o" "sketch\\src\\HAL\\shared\\backtrace\\unwarmmem.cpp.o" "sketch\\src\\HAL\\shared\\backtrace\\unwinder.cpp.o" "sketch\\src\\HAL\\shared\\backtrace\\unwmemaccess.cpp.o" "sketch\\src\\HAL\\shared\\eeprom_i2c.cpp.o" "sketch\\src\\HAL\\shared\\eeprom_spi.cpp.o" "sketch\\src\\HAL\\shared\\persistent_store_api.cpp.o" "sketch\\src\\HAL\\shared\\servo.cpp.o" "sketch\\src\\MarlinCore.cpp.o" "sketch\\src\\core\\multi_language.cpp.o" "sketch\\src\\core\\serial.cpp.o" "sketch\\src\\core\\utility.cpp.o" "sketch\\src\\feature\\I2CPositionEncoder.cpp.o" "sketch\\src\\feature\\Max7219_Debug_LEDs.cpp.o" "sketch\\src\\feature\\babystep.cpp.o" "sketch\\src\\feature\\backlash.cpp.o" "sketch\\src\\feature\\baricuda.cpp.o" "sketch\\src\\feature\\bedlevel\\abl\\abl.cpp.o" "sketch\\src\\feature\\bedlevel\\bedlevel.cpp.o" "sketch\\src\\feature\\bedlevel\\mbl\\mesh_bed_leveling.cpp.o" "sketch\\src\\feature\\bedlevel\\ubl\\ubl.cpp.o" "sketch\\src\\feature\\bedlevel\\ubl\\ubl_G29.cpp.o" "sketch\\src\\feature\\bedlevel\\ubl\\ubl_motion.cpp.o" "sketch\\src\\feature\\binary_protocol.cpp.o" "sketch\\src\\feature\\bltouch.cpp.o" "sketch\\src\\feature\\cancel_object.cpp.o" "sketch\\src\\feature\\caselight.cpp.o" "sketch\\src\\feature\\closedloop.cpp.o" "sketch\\src\\feature\\controllerfan.cpp.o" "sketch\\src\\feature\\dac\\dac_dac084s085.cpp.o" "sketch\\src\\feature\\dac\\dac_mcp4728.cpp.o" "sketch\\src\\feature\\dac\\stepper_dac.cpp.o" "sketch\\src\\feature\\digipot\\digipot_mcp4018.cpp.o" "sketch\\src\\feature\\digipot\\digipot_mcp4451.cpp.o" "sketch\\src\\feature\\emergency_parser.cpp.o" "sketch\\src\\feature\\fanmux.cpp.o" "sketch\\src\\feature\\filwidth.cpp.o" "sketch\\src\\feature\\fwretract.cpp.o" "sketch\\src\\feature\\host_actions.cpp.o" "sketch\\src\\feature\\joystick.cpp.o" "sketch\\src\\feature\\leds\\blinkm.cpp.o" "sketch\\src\\feature\\leds\\leds.cpp.o" "sketch\\src\\feature\\leds\\neopixel.cpp.o" "sketch\\src\\feature\\leds\\pca9632.cpp.o" "sketch\\src\\feature\\leds\\printer_event_leds.cpp.o" "sketch\\src\\feature\\leds\\tempstat.cpp.o" "sketch\\src\\feature\\mixing.cpp.o" "sketch\\src\\feature\\pause.cpp.o" "sketch\\src\\feature\\power.cpp.o" "sketch\\src\\feature\\power_loss_recovery.cpp.o" "sketch\\src\\feature\\probe_temp_compensation.cpp.o" "sketch\\src\\feature\\prusa_MMU2\\mmu2.cpp.o" "sketch\\src\\feature\\runout.cpp.o" "sketch\\src\\feature\\snmm.cpp.o" "sketch\\src\\feature\\solenoid.cpp.o" "sketch\\src\\feature\\spindle_laser.cpp.o" "sketch\\src\\feature\\tmc_util.cpp.o" "sketch\\src\\feature\\touch\\xpt2046.cpp.o" "sketch\\src\\feature\\twibus.cpp.o" "sketch\\src\\feature\\z_stepper_align.cpp.o" "sketch\\src\\gcode\\bedlevel\\G26.cpp.o" "sketch\\src\\gcode\\bedlevel\\G42.cpp.o" "sketch\\src\\gcode\\bedlevel\\M420.cpp.o" "sketch\\src\\gcode\\bedlevel\\abl\\G29.cpp.o" "sketch\\src\\gcode\\bedlevel\\abl\\M421.cpp.o" "sketch\\src\\gcode\\bedlevel\\mbl\\G29.cpp.o" "sketch\\src\\gcode\\bedlevel\\mbl\\M421.cpp.o" "sketch\\src\\gcode\\bedlevel\\ubl\\G29.cpp.o" "sketch\\src\\gcode\\bedlevel\\ubl\\M421.cpp.o" "sketch\\src\\gcode\\calibrate\\G28.cpp.o" "sketch\\src\\gcode\\calibrate\\G33.cpp.o" "sketch\\src\\gcode\\calibrate\\G34_M422.cpp.o" "sketch\\src\\gcode\\calibrate\\G425.cpp.o" "sketch\\src\\gcode\\calibrate\\G76_M871.cpp.o" "sketch\\src\\gcode\\calibrate\\M100.cpp.o" "sketch\\src\\gcode\\calibrate\\M12.cpp.o" "sketch\\src\\gcode\\calibrate\\M425.cpp.o" "sketch\\src\\gcode\\calibrate\\M48.cpp.o" "sketch\\src\\gcode\\calibrate\\M665.cpp.o" "sketch\\src\\gcode\\calibrate\\M666.cpp.o" "sketch\\src\\gcode\\calibrate\\M852.cpp.o" "sketch\\src\\gcode\\config\\M200-M205.cpp.o" "sketch\\src\\gcode\\config\\M217.cpp.o" "sketch\\src\\gcode\\config\\M218.cpp.o" "sketch\\src\\gcode\\config\\M220.cpp.o" "sketch\\src\\gcode\\config\\M221.cpp.o" "sketch\\src\\gcode\\config\\M281.cpp.o" "sketch\\src\\gcode\\config\\M301.cpp.o" "sketch\\src\\gcode\\config\\M302.cpp.o" "sketch\\src\\gcode\\config\\M304.cpp.o" "sketch\\src\\gcode\\config\\M305.cpp.o" "sketch\\src\\gcode\\config\\M43.cpp.o" "sketch\\src\\gcode\\config\\M540.cpp.o" "sketch\\src\\gcode\\config\\M575.cpp.o" "sketch\\src\\gcode\\config\\M672.cpp.o" "sketch\\src\\gcode\\config\\M92.cpp.o" "sketch\\src\\gcode\\control\\M108_M112_M410.cpp.o" "sketch\\src\\gcode\\control\\M111.cpp.o" "sketch\\src\\gcode\\control\\M120_M121.cpp.o" "sketch\\src\\gcode\\control\\M17_M18_M84.cpp.o" "sketch\\src\\gcode\\control\\M211.cpp.o" "sketch\\src\\gcode\\control\\M226.cpp.o" "sketch\\src\\gcode\\control\\M280.cpp.o" "sketch\\src\\gcode\\control\\M3-M5.cpp.o" "sketch\\src\\gcode\\control\\M350_M351.cpp.o" "sketch\\src\\gcode\\control\\M380_M381.cpp.o" "sketch\\src\\gcode\\control\\M400.cpp.o" "sketch\\src\\gcode\\control\\M42.cpp.o" "sketch\\src\\gcode\\control\\M605.cpp.o" "sketch\\src\\gcode\\control\\M7-M9.cpp.o" "sketch\\src\\gcode\\control\\M80_M81.cpp.o" "sketch\\src\\gcode\\control\\M85.cpp.o" "sketch\\src\\gcode\\control\\M997.cpp.o" "sketch\\src\\gcode\\control\\M999.cpp.o" "sketch\\src\\gcode\\control\\T.cpp.o" "sketch\\src\\gcode\\eeprom\\M500-M504.cpp.o" "sketch\\src\\gcode\\feature\\L6470\\M122.cpp.o" "sketch\\src\\gcode\\feature\\L6470\\M906.cpp.o" "sketch\\src\\gcode\\feature\\L6470\\M916-918.cpp.o" "sketch\\src\\gcode\\feature\\advance\\M900.cpp.o" "sketch\\src\\gcode\\feature\\baricuda\\M126-M129.cpp.o" "sketch\\src\\gcode\\feature\\camera\\M240.cpp.o" "sketch\\src\\gcode\\feature\\cancel\\M486.cpp.o" "sketch\\src\\gcode\\feature\\caselight\\M355.cpp.o" "sketch\\src\\gcode\\feature\\clean\\G12.cpp.o" "sketch\\src\\gcode\\feature\\digipot\\M907-M910.cpp.o" "sketch\\src\\gcode\\feature\\filwidth\\M404-M407.cpp.o" "sketch\\src\\gcode\\feature\\fwretract\\G10_G11.cpp.o" "sketch\\src\\gcode\\feature\\fwretract\\M207-M209.cpp.o" "sketch\\src\\gcode\\feature\\i2c\\M260_M261.cpp.o" "sketch\\src\\gcode\\feature\\leds\\M150.cpp.o" "sketch\\src\\gcode\\feature\\leds\\M7219.cpp.o" "sketch\\src\\gcode\\feature\\macro\\M810-M819.cpp.o" "sketch\\src\\gcode\\feature\\mixing\\M163-M165.cpp.o" "sketch\\src\\gcode\\feature\\mixing\\M166.cpp.o" "sketch\\src\\gcode\\feature\\pause\\G27.cpp.o" "sketch\\src\\gcode\\feature\\pause\\G60.cpp.o" "sketch\\src\\gcode\\feature\\pause\\G61.cpp.o" "sketch\\src\\gcode\\feature\\pause\\M125.cpp.o" "sketch\\src\\gcode\\feature\\pause\\M600.cpp.o" "sketch\\src\\gcode\\feature\\pause\\M603.cpp.o" "sketch\\src\\gcode\\feature\\pause\\M701_M702.cpp.o" "sketch\\src\\gcode\\feature\\powerloss\\M1000.cpp.o" "sketch\\src\\gcode\\feature\\powerloss\\M413.cpp.o" "sketch\\src\\gcode\\ [... cut ...]
Bibliothek SrcWrapper in Version 1.0.1 im Ordner: C:\Users\s\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.8.0\libraries\SrcWrapper  wird verwendet
Bibliothek SPI in Version 1.0 im Ordner: C:\Users\s\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.8.0\libraries\SPI  wird verwendet
Bibliothek IWatchdog in Version 1.0.0 im Ordner: C:\Users\s\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.8.0\libraries\IWatchdog  wird verwendet
exit status 1
Fehler beim Kompilieren für das Board 3D printer boards. 
```

rennwaffen, 2020-02-21:
I'm getting an error with the M200 build as well, but in my case it may have something to do with the platform.ino file:

[Configuration.txt](https://github.com/MarlinFirmware/Configurations/files/4237746/Configuration.txt)
[Configuration.zip](https://github.com/MarlinFirmware/Configurations/files/4237748/Configuration.zip)

[platformio]
src_dir = Marlin
boards_dir = buildroot/share/PlatformIO/boards
default_envs = STM32F103CB_malyan

I'm trying to compile for a Malyan M300/Monoprice Mini Delta - which uses a board similar to the MP Mini Select/Malyan M200 however I'm getting an error when compiling:
C:\users[USERNAME].platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple\HardwareTimer.cpp:32:10: fatal error: board/board.h: No such file or directory

freiform, 2020-02-26:
Putting an empty implementation of `ExtUI::OnPidTuning(ExtUI::result_t)` into `extui_malayan_lcd.cpp fixes the error and the build runs though. Whether or not this is feasible, I do not know.

freiform, 2020-03-01:
With the empty implementetion it builds, but doesn't run. The printer is continuously resetting. It seems Marlin is broken for the M200.

thinkyhead, 2020-03-14:
This may now be working. Please test the latest version.

freiform, 2020-03-18:
arduino-builder now throws error 137 (that I cannot find any information on) and exits.

thinkyhead, 2020-03-25:
Try deleting all the HAL sub-folders except for the `STM32F1` and `shared` folders to eliminate any issues with over-long command lines. This will also speed up the build.

We're also looking at getting Malyan M200 v1 and v2 building properly within PlatformIO over at https://community.platformio.org/t/marlin-compile-fail-board-board-h-no-such-file-or-directory/11843/12 and that will be the better option going forward once it's working….

thinkyhead, 2020-03-31:
Discussion can continue at MarlinFirmware/Marlin#17352

closed 2020-03-31 by thinkyhead