MarlinFirmware/Configurations #22: Creality Cr-10 and cr-10S bed wrong configs

Liger0, 2020-02-22:
### Bug Description
The creality cr-10 and cr-10s bed related settings are wrong (probably the cr-10s4 and cr-10s5 too since they share the same bed offsets and have the exact same heater, but a bigger and heat-dispersive aluminium)

For the heating feature, these are the stock codes in the firmware:

 ```
 //120V 250W silicone heater into 4mm borosilicate (MendelMax 1.5+)
  //from FOPDT model - kp=.39 Tp=405 Tdead=66, Tc set to 79.2, aggressive factor of .15 (vs .1, 1, 10)
  #define DEFAULT_bedKp 10.00
  #define DEFAULT_bedKi .023
  #define DEFAULT_bedKd 305.4

  //120V 250W silicone heater into 4mm borosilicate (MendelMax 1.5+)
  //from pidautotune
  //#define DEFAULT_bedKp 97.1
  //#define DEFAULT_bedKi 1.41
  //#define DEFAULT_bedKd 1675.16

```

those are unrelated from the cr-10(S) and don't work.
No matter if PID is enabled or disabled, the heating fails anyway with those values.
If I use the one defined there without manual PID, I can't print over 60°C, which even takes an eternity to reach.
If I define the manual PID autotune, the PID autotuning feature doesn't work because it still uses the defined PID values as reference, so there is no way around that and still fails the PID autotuning at 60°C.

I worked manually on them until being able to do a proper PID autotune at 80°C (mid value between 60°C for PLA and 100°C for ABS), and this is the PID value that results:

```
  //Stock cr10s bed and stock 4mm borosilicate glass
  #define DEFAULT_bedKp 185.12
  #define DEFAULT_bedKi 36.99
  #define DEFAULT_bedKd 617.69
```

with this I can easily print even on ABS temp at 110°C.

Talking about mechanical limits, the bed area settings don't make much sense.

This setting is non-sense:

```
#define X_BED_SIZE 296 // Nozzle is at X4 when homed
#define Y_BED_SIZE 298 // Nozzle is at Y2 when homed
```
it should be, instead

```
#define X_BED_SIZE 300 // Nozzle is at X0 when homed
#define Y_BED_SIZE 300 // Nozzle is at Y0 when homed
```

again, this doesn't make sense
```
#define X_MAX_POS 319 // 15mm Extra space on the right end
#define Y_MAX_POS 306 // 4mm Extra space behind the bed
```

what is correct is, since the bed size is 5mm ahead:
```
#define X_MAX_POS 305 // 5mm Extra space on the right end
#define Y_MAX_POS 305 // 5mm Extra space behind the bed
```
same for this, which is wrong:
```
#define MANUAL_X_HOME_POS 4
#define MANUAL_Y_HOME_POS 2
```
while the correct values are

```
//#define MANUAL_X_HOME_POS 0
//#define MANUAL_Y_HOME_POS 0
```
because the bed is larger of 5mm on the negative side, so printing at the homing position is not on the edge and is totally fair.

Deneteus, 2020-03-16:
I had the same heating problem with the CR-10 (300) and kept being told either my beds were bad (3 separate printers) or they needed to be insulated and I installed insulation and still had problems. 

I tried with or without PIDTEMPBED and I couldn't heat past 70C so I just changed the thermal protection timeouts til I could make it to at least 75C. With Marlin 2.0 I couldn't get any higher. I ran PID Autotune on 3 printers at the same time to see what kind of PID values I was getting. 

In the stock firmware PIDTEMPBED is **disabled.** The original CR-10 bed's could never make it to 100-110C even with the original firmware. It did heat to 90C and it didn't take a super long time either to get to 75-80C.

I never saw the changes that were made to the bed size. Have you compared them with the original stock firmware configs?
https://www.creality3dofficial.com/pages/firmware-download

Now that the configuration files are no longer in zip format I suggest comparing the old configs to the new ones.

thinkyhead, 2020-03-28:
The values found in the example configurations today are based on the best information and corrections we have received from users over the years (and our own preferred settings for the CR-10S machines we test on).

Rather than change the configs here and there and now and then, according to occasional user comments, it would be better if some Creality3D fans took on the job of maintaining the configurations and kept them up to date.

Everyone who contributes to the project brings their own knowledge, and it's not reliable to ask someone else to make the changes. There's no telling when those changes will be made.

Liger0, 2020-03-28:
So basically no one will be able to print anything that isn't PLA until somehow the wrong PID get removed for correct ones.

thinkyhead, 2020-03-31:
[Somehow.](http://marlinfw.org/docs/development/getting_started_pull_requests.html)

swilkens, 2020-04-20:
@thinkyhead @thisiskeithb from the top of my head, the only time we actively diverted from settings sourced from Creality sources was when we slowed down the homing speeds due to the end stops bottoming out. For the rest, I think it's fair to say the configs here remained fairly close to the available Creality sources.
 
I could go over the sources Creality offers and check the configs in our repo against the values Creality provides, create a PR to bring our configs to Creality config values (for things like bed size, pids, speeds.. etc etc etc)

Opinions on this approach? The work is rather trivial, but not starting it if you know of a valid objection.

thisiskeithb, 2020-04-20:
> I could go over the sources Creality offers and check the configs in our repo against the values Creality provides, create a PR to bring our configs to Creality config values (for things like bed size, pids, speeds.. etc etc etc). Maybe add printers where missing.

You have to be careful doing this. For example, the Ender-5 config from Creality has defaults of homing to X & Y min instead of X & Y max (the endstops are in the back right corner on the physically machine.

Creality also *still* ships without thermal runaway protection enabled.

We could go back and forth fighting over which source is "correct" forever.

Liger0, 2020-04-20:
Correcting the PID alone would be enough to let people use the machine.
As of now you can only print PLA (and it takes an eternity because of how the configs are wrong)

swilkens, 2020-04-20:
@thisiskeithb we could start by generating an inventory of differences? Obviously leaving things like thermal management off would be something we can filter out, but different pids and bed sizes from readily available sources seems like a glaring oversight that could be quickly fixed by a review like this.

thinkyhead, 2020-04-23:
> Correcting the PID alone would be enough

People should always expect our examples to be likely different from your actual hardware. Your own hotend or room temperature may not be anything like the one the original tester used. And the MCU speed and the particular resistances and quirks of your board can also matter. So always run `M303` for your hotend and bed after any firmware update.

If someone has a good middle-of-the-road PID setting that works for a wide set of cases, submit that.

Liger0, 2020-04-23:
I don't know how much times I said this but M303 does not work if the PID is too different from the actual hardware, it will fail at 55°C.
And the hardware used to calculate the current PID is not from a cr10S, it's way too different.
If anyone is able to set a middle PID I provided mine, and possibly fixing the PID autotune to work without considering the already set PID would be good.

thinkyhead, 2020-04-27:
> M303 does not work if the PID is too different from the actual hardware

Yes, this is well understood, and YouTube tutorials on doing `M303` tuning should tell you to modify or hand tune your PID values in that case before attempting the next `M303`. They may also suggest hand-tuning as a followup to the rough work done by `M303`, to get that last bit of wobble out.

While this discussion is interesting, I would personally prefer that people "adopt" configurations and become the caretakers of those configurations for their favorite machines so that I don't have to think about them. 🧠  None of the maintainers enjoys being told what changes we oughta make (when we find the time) when the person explaining it could submit those changes themselves. And if you want something done right…

Ultimately it would be nifty to get `M303` doing all the magic for you, but no one in my social circles has the time with all the other things that there are to fix right now….

richsadler, 2021-02-10:
Has anyone ever been able to fix the problem? I bought a CR10 a couple of months ago and ultimately upgraded the mother board and firmware to Marlin 2.0.7.2, and all of a sudden I've ran into this issue too, and I can totally understand the frustration. The printer is pretty much useless now, it just amazes me that this is an issue for so many people (from what I've seen during all of my rearching for a solution) and it's existed for so long.

steckdev, 2021-02-18:
I still haven't been able to resolve this issue. I have a CR-10 that I am attempting to heat the printer bed. However it will only heat to 10' shy of the target temperature. Then it just holds and won't heat anymore. (For example only heats to 50 instead of 60...) Anyone have a guess of what I am doing wrong?

Liger0, 2021-02-19:
> I still haven't been able to resolve this issue. I have a CR-10 that I am attempting to heat the printer bed. However it will only heat to 10' shy of the target temperature. Then it just holds and [...]

Use the pid values I posted and enable pid bed.

Deneteus, 2021-02-19:
> Has anyone ever been able to fix the problem? I bought a CR10 a couple of months ago and ultimately upgraded the mother board and firmware to Marlin 2.0.7.2, and all of a sudden I've ran into this [...]

No, It was never resolved.  There was a change that broke the heating from the stock firmware. Trying to PID autotune causes it to eventually fail on the stock CR-10. I even went and added insulation.

Liger0, 2021-02-19:
> 
> 
> > Has anyone ever been able to fix the problem? I bought a CR10 a couple of months ago and ultimately upgraded the mother board and firmware to Marlin 2.0.7.2, and all of a sudden I've ran into this issue too, and I can totally understand the frustration. The printer is pretty much useless now, it just amazes me that this is an issue for so many people (from what I've seen during all of my rearching for a solution) and it's existed for so long.
> 
> @richsadler @Liger0
> 
> No, It was never resolved. There was a change that broke the heating from the stock firmware. Trying to PID autotune causes it to eventually fail on the stock CR-10 with no modifications and stock SSR. I even went and added insulation. I ran PID Autotune. I tested it with the bed heated and unheated. I tried changing the temp/time values for the protection. It just times out like its stalled and I get the beeping of doom. I have not tested it in current Marlin builds yet. I took a break and worked on other stuff where I knew it did work. I bought new SKR boards to replace the stock ones. I have other printers and it works fine for them.
> 
> @thinkyhead I wish I had more time to dedicate to making changes to Marlin. Houston, Texas is in a bit of a crunch right now. :) I'll test anything you throw my way though.

As I just said, use my pid values and do pid autotune.

Deneteus, 2021-02-19:
What version of Marlin had you tested this under?

Liger0, 2021-02-20:
A november bugfix after the 2.0.7.2 stable release.

Deneteus, 2021-03-19:
> A november bugfix after the 2.0.7.2 stable release.

I took the stock config and added your PID settings and it still failed.

NapAndGrit, 2021-03-19:
> I still haven't been able to resolve this issue. I have a CR-10 that I am attempting to heat the printer bed. However it will only heat to 10' shy of the target temperature. Then it just holds and [...]

I had the same problem with stock FW and Unified but I was able to eventually fix it on Unified.  I really wanted to use manual mesh leveling so I was going to flash the latest Marlin 2 and something told me just to check the pull requests, I'm glad I did.  So if I were to flash Marlin 2 I would have the same problem?

Liger0, 2021-03-19:
> 
> 
> > A november bugfix after the 2.0.7.2 stable release.

after inserting my values, did you enable bed PID instead of the BANG option, and then do a PID autotune?

NapAndGrit, 2021-03-20:
If you're asking me, no I didn't.  I still had TH3D when I commented but I went ahead and installed Marlin 2 today and it was pretty squirrelly at first but after a few auto tunes it seems to be doing fine if I don't go crazy on the speed.  I tried some of the experimental features like the fan power scaling and the extrusion scaling or whatever its called but I still haven't calibrated for them yet.  I have had a few runaways in the 70's though so obviously its still not working as it should.  With TH3D I was printing a 1.0mm nozzle and with linear advance I could do infill (not solid) at 100 mm/s or more on some things.     As soon as my print finishes I'll try your values just for the sake finding out what happens and let you know if it works.

NapAndGrit, 2021-03-20:
I just realized none of what I said had anything to do with the bed... oops.  I've been watching the hot end temp because that's what keeps triggering runaways, I'm printing PLA and PETG so its not going above 70 anyway.

mydevpeeps, 2021-07-12:
I will be closely watching this thread as I have the same issue right now on a brand new CR10S-S5 with the newer board, the stock 1.1.6 that I can download from creality will run a job, heat up the bed, and do it's thing. When i compiled 1.1.8 for bltouch i got everything working, but the bed now won't heat. It did with the stock 1.1.6. I end up preheating using the controller and then fire off my print jobs to get them to run. I'm not even using PID mode.. it's still set on bangbang full current mode and it still errors out. Duet has a timeout value that can be set to force a delay for slow beds and I wish I had this here so I could just set the timeout to be longer and wait it out.

Deneteus, 2021-07-12:
@mydevpeeps I would just upgrade to a current Marlin 2.0.9.1 build and not even bother with Marlin 1.1.8.    
I replaced all of my boards with SKR E3 DIP v1.1 and went about my merry way. All my beds heat up with no issues.

mydevpeeps, 2021-09-20:
> @mydevpeeps I would just upgrade to a current Marlin 2.0.9.1 build and not even bother with Marlin 1.1.8.
> I replaced all of my boards with SKR E3 DIP v1.1 and went about my merry way. All my beds heat up with no issues.

I have upgraded to 2.0.9.1-bugfix as well. Thanks for the suggestion.

Deneteus, 2021-09-22:
@thinkyhead You can close this. Thanks. :)

closed 2021-10-30 by thinkyhead

CNCXPROUK, 2022-07-05:
I just want to add a note. This has never been fixed for me in any marlin release. My CR-10s is one of the earlier versions and the 12v psi just could never heat is fast enough. Thats until I use @Liger0 s pid settings.  One issue was that despite refreshing the firmware and resetting the EEPROM the settings weren't loaded into the EEPROM. I manually put them in to the EEPROM and tried a PID tune at 80, then 90 degrees and its worked. This printer has never been able to get above 90 degrees and always failed with a time out trying to reach the temperature.

leodadcyclizt, 2023-03-04:
@thinkyhead @Liger0 

I found this thread after experiencing the same slow bed heating/failure issue with CR-10S after I upgraded from Marlin 1.1.6 to 2.1.2 - I know this issue is 'closed,' but I wanted to at least take a shot at fixing it before I surrender to buying an SKR Mini card as others have suggested. Full transparency - I'm an amateur at best with configuring or even understanding Marlin - but I considered what someone else posted about taking inventory of differences between earlier and current versions. 
I compared Creality's source code for my printer that I used when I installed a BLTouch (the last time my bed temp worked fine) versus the newer 2.1.2. 
- 2.1.2 version was set to "1 : 100kΩ  EPCOS - Best choice for EPCOS thermistors" 
- 1.1.6 version was set to "5 : 100K thermistor - ATC Semitec 104GT-2 (Used in ParCan & J-Head) (4.7k pullup)"

They're both 100kohm but the thermistor tables for each are pretty different - could this significantly affect my bed heating performance? I haven't changed any heating elements or thermistors since purchasing the machine.
[image]