MarlinFirmware/Configurations #23: Delta auto calibration failing

GravityWinz, 2020-02-23:
I had been running 2.0.0 using the Example Anycubic Kossel configs modified for Bigtree SKR V1_3, and TMC2208 drivers. But with 2.0.4, when I attempt to perform a Delta Auto Calibration, I am prompted to deploy my probe (which I do) the machine probes a single point, it requests that I stow the probe and then issues a probing failed error message. 

I built using as vanilla 2.0.4 configuration as possible, starting with the 2.0.4 config releases of the examples/delta/Anycubic/Kossel config files, and have just made the minimal changes for my board and the TMC2208 drivers. For reference I have attached the config files here: 
[Configuration_files.zip](https://github.com/MarlinFirmware/Configurations/files/4242303/Configuration_files.zip)

Burnallover, 2020-02-24:
Same for me, with BTT 1.4 Turbo, Marlin 2.0.3 working, but with marlin 2.0.4 i can't do an auto calibration or an Auto Bed Leveling.

redkuul, 2020-02-25:
Same for me, with BTT SKR 1.3 and TMC 2209.

Schild0r, 2020-03-08:
Not sure whether this works for everyone but this is what it fixed this issue for me:
- Home printer
- move nozzle as close to the bed as possible
- note what z height is displayed and substract this from Delta_height (maybe add .5mm)
- reset Ex,Ey,Ez,Tx,Ty,Tz to 0
- recompile, init eeprom
-rerun delta calibration

thinkyhead, 2020-03-25:
This Issue Queue is for bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose) to the main Marlin repo, because it could be a bug in the code.

closed 2020-03-25 by thinkyhead

Sonsi79, 2020-03-25:
Same Error for me with Anycubic i3 Mega with BL-Touch.
Bugfix.2.0.X works great but since 2.0.4.4 still error. Settings seems to be right :-(

thinkyhead, 2020-03-31:
> Bugfix.2.0.X works great 

Good to hear the bug is resolved in the latest code.