MarlinFirmware/Configurations #24: Skr 1.3 + TMC2130 + TFT35 v2 - Steppers do not work

mxmxx93, 2020-02-24:
Hi everyone, 

first of all I wanted to apologize if there are errors in the post but I'm new, I used the search button but nothing about this configuration.

My configuration is the one mentioned in the title. My problem is that once I launch the command (both from pc and TFT) to move the steppers they do not respond and do not move.

I tried to mount several Firmware from 2.0.0 to 2.0.4, but the problem remains the same. I followed the guide ( link ) to set the firmware and everything, but the result was always negative. 

if you need any more information, I will gladly provide it, I would be very grateful. Unfortunately, my printer is OFF since Christmas for this reason... and now I'm giving up and thinking of giving up

DaSTIG, 2020-03-12:
I would start by enabling TMC_Debug in firmware, and getting the  stepper drivers status with M122.  If you have the firmware not configured correctly, they arent going to move.   I been using the SKR 1.3 with 2130s for 8 months, and they work great.

oldman4U, 2020-03-13:
Hi.

Has the printer ever worked before and what changed then?
Could you attach some pictures of the mainboard and the drivers and some more information about the mode you are using.

The (link) does not work.

thinkyhead, 2020-03-31:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose) to the main project.

closed 2020-03-31 by thinkyhead