MarlinFirmware/Configurations #25: Cr-10 default display not showing anything

thomaswang04, 2020-02-24:
This also happened with 1.1.9 and I thought the newer firmware would fix it but it hasn't. Marlin 2.0 is loaded and I am able to use it in serial monitor however, nothing shows up on the screen. Its just a blank screen that is lit up .

thinkyhead, 2020-03-25:
But the screen works with 1.1.8?

thinkyhead, 2020-03-25:
This repo is still a work in progress and as we are under-staffed, it isn't getting as much attention as it might. For best results getting help with configuration and troubleshooting try the following resources first:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose) to the main Marlin repo, since it could be an issue in the code.

closed 2020-03-25 by thinkyhead