MarlinFirmware/Configurations #26: lcd sd card does not work

shawnp30, 2020-02-25:
everything compiles for btt gtr v 1.0. when i try and use sd card slot on tft35 v3. nothing shows up. if i take the card out of the main board and replace it with an sd card with an stl file on it the gtr board reads the card. why is the card not being read by the tft35? i have sd card enabled in config. I noticed when i went into pins i saw an option for custom spi pins. once i disabled that feature the lcd was able to read the card, but i cannot print. When i try and select an stl and click on it, it goes to the main screen.

AnHardt, 2020-02-25:
> When i try and select an stl and click on it ...

Marlin is not a slicer. It dos not now what to do with an "\*.stl"-file.
Marlin can open g-code-files with the endings "\*.g\*"
The displays firmware is more restricted. It can't see files different from "\*.gcode".

shawnp30, 2020-02-25:
*fixed*

closed 2020-03-25 by thinkyhead