MarlinFirmware/Configurations #27: TMC2208 "TX only" does not compile in 2.0.x but works perfectly in 1.1.x

Bugg69s, 2020-02-29:
**marlin doesn't compile if you set tmc2208 for TX only (ender 3 melzi) on marlin 2.0.x but does on 1.1.x
like this:
#define Y_SERIAL_TX_PIN 14
#define Y_SERIAL_RX_PIN -1

if you do it on 2.0.x it errors out like this:**
============================================================
sketch\src\inc/SanityCheck.h:2088:4: error: #error "TMC2208 or TMC2209 on Y requires Y_HARDWARE_SERIAL or Y_SERIAL_(RX|TX)_PIN."

   #error "TMC2208 or TMC2209 on Y requires Y_HARDWARE_SERIAL or Y_SERIAL_(RX|TX)_PIN."

    ^~~~~
=============================================================

**And after disabling sanity check it errors out like this:**
=============================================================
fork/exec C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\hardware\tools\avr/bin/avr-gcc.exe: The filename or extension is too long.
Error compiling for board Sanguino.
=============================================================

**I have tried with my setup and stock setup and just adding those two lines and just selected tmc2208 stepper for Y stepper driver**

**my marlin setup:** https://drive.google.com/open?id=1hROjDBbQuNfpImhbR3PDjXNqvJSD0zu0

AnHardt, 2020-02-29:
Please try:
```
#define Y_SERIAL_TX_PIN 14
#define Y_SERIAL_RX_PIN 14
```

Bugg69s, 2020-02-29:
@AnHardt no luck, it complains about:
avr-gcc: error: CreateProcess: No such file or directory

Bugg69s, 2020-03-03:
> Please try:
> 
> ```
> #define Y_SERIAL_TX_PIN 14
> #define Y_SERIAL_RX_PIN 14
> ```

Specifically, something in the TMC lib. Because when you use the other TMC library it spits out the exact same error but with little bit different expression

thinkyhead, 2020-03-25:
> avr-gcc: error: CreateProcess: No such file or directory

Re-install PlatformIO and try again. The easiest option is [described here](https://marlinfw.org/docs/basics/auto_build_marlin.html).

Rubenazzzo, 2020-05-06:
HI folks.
I'm sorry if it's not the right place, but I take advantage of the thread.
Today configuring the TMC2208 UART in my re-arm, I have realized 2 things:
1-before activating the SD in the panel, it did not give me the error message of "TMC ERROR", and when I have activated it, it shows me the message (I have not tried again), but I have seen that they work, physically, I have not tried pronterface.
2- in the "pins" tab in the area that indicates the pins for TX and RX, I get dark options, when normally, if everything is related to the marlin in VSC, it is highlighted.
#if HAS_TMC_UART
  / **
   * TMC2208 / TMC2209 stepper drivers
   *
   * Hardware serial communication ports.
   * If undefined software serial is used according to the pins below
   * /
  // # define X_HARDWARE_SERIAL Serial1
  // # define X2_HARDWARE_SERIAL Serial1
  // # define Y_HARDWARE_SERIAL Serial1
  // # define Y2_HARDWARE_SERIAL Serial1
  // # define Z_HARDWARE_SERIAL Serial1
  // # define Z2_HARDWARE_SERIAL Serial1
  // # define E0_HARDWARE_SERIAL Serial1
  // # define E1_HARDWARE_SERIAL Serial1
  // # define E2_HARDWARE_SERIAL Serial1
  // # define E3_HARDWARE_SERIAL Serial1
  // # define E4_HARDWARE_SERIAL Serial1

  //
  // serial software
  //

  #ifndef X_SERIAL_TX_PIN
    #define X_SERIAL_TX_PIN 40
  #endif
  #ifndef X_SERIAL_RX_PIN
    #define X_SERIAL_RX_PIN 63
  #endif
  #ifndef X2_SERIAL_TX_PIN
    #define X2_SERIAL_TX_PIN -1
  #endif
  #ifndef X2_SERIAL_RX_PIN
    #define X2_SERIAL_RX_PIN -1
  #endif

  #ifndef Y_SERIAL_TX_PIN
    #define Y_SERIAL_TX_PIN 59
  #endif
  #ifndef Y_SERIAL_RX_PIN
    #define Y_SERIAL_RX_PIN 64
  #endif
  #ifndef Y2_SERIAL_TX_PIN
    #define Y2_SERIAL_TX_PIN -1
  #endif
  #ifndef Y2_SERIAL_RX_PIN
    #define Y2_SERIAL_RX_PIN -1

Rubenazzzo, 2020-05-06:
sorry, I use  2.0.5.3 version

thinkyhead, 2020-05-07:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

Rubenazzzo, 2020-05-14:
Hi, if is for me ask, can close this problem, finally install a skr 1.3, with 2209 and now have others problems, but i,ll open others issues.
Thanks for your time.

Bugg69s, 2020-05-14:
> Hi, if is for me ask, can close this problem, finally install a skr 1.3, with 2209 and now have others problems, but i,ll open others issues.
> Thanks for your time.

You might try marlin discord channel

closed 2020-05-14 by Bugg69s