MarlinFirmware/Configurations #31: Creality Marlin 5 configuration not working with Marlin 2.0.4.4

patoberli, 2020-03-10:
Hi all
I'm trying to compile Marlin 2.0.4.4 for a Creality Ender-5 with the Bigtreetech SKR Mini 1.2 mainboard. I've taken the configurations from here and modified the needed values for the different board.

VSCode complains this:
`The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

> Executing task: platformio run <

Processing STM32F103RC_bigtree (platform: ststm32; board: genericSTM32F103RC; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RC.html
PLATFORM: ST STM32 6.0.0 > STM32F103RC (48k RAM. 256k Flash)
HARDWARE: STM32F103RCT6 72MHz, 48KB RAM, 256KB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES: 
 - framework-arduinoststm32-maple 1.10000.190819 (1.0.0) 
 - tool-stm32duino 1.0.1 
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 38 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <LiquidCrystal>
|-- <TMCStepper> 0.6.2
|   |-- <SoftwareSerialM> 1.0.0
|-- <U8glib-HAL> 0.4.1
|   |-- <Wire> 1.0
|-- <Adafruit MAX31865 library> 1.0.3
|-- <LiquidTWI2>
|   |-- <Wire> 1.0
|-- <Arduino-L6470> 0.8.0
|-- <SailfishLCD>
|-- <SailfishRGB_LED>
|   |-- <Wire> 1.0
|-- <SlowSoftI2CMaster>
|-- <SoftwareSerialM> 1.0.0
|-- <STM32ADC> 1.0
|-- <USBComposite for STM32F1> 0.91
|-- <EEPROM>
|-- <Wire> 1.0
Building in release mode
Compiling .pio/build/STM32F103RC_bigtree/src/src/HAL/HAL_STM32F1/HAL.cpp.o
Compiling .pio/build/STM32F103RC_bigtree/src/src/HAL/HAL_STM32F1/HAL_SPI.cpp.o
Compiling .pio/build/STM32F103RC_bigtree/src/src/HAL/HAL_STM32F1/Servo.cpp.o
In file included from Marlin/src/HAL/HAL_STM32F1/../../inc/MarlinConfig.h:38:0,
                 from Marlin/src/HAL/HAL_STM32F1/HAL.cpp:30:
Marlin/src/HAL/HAL_STM32F1/../../inc/SanityCheck.h:45:4: error: #error "You are using an old Configuration.h file, update it before building Marlin."
   #error "You are using an old Configuration.h file, update it before building Marlin."
    ^~~~~
In file included from Marlin/src/HAL/HAL_STM32F1/../../inc/MarlinConfig.h:38:0,
                 from Marlin/src/HAL/HAL_STM32F1/HAL.cpp:30:
Marlin/src/HAL/HAL_STM32F1/../../inc/SanityCheck.h:49:4: error: #error "You are using an old Configuration_adv.h file, update it before building Marlin."
   #error "You are using an old Configuration_adv.h file, update it before building Marlin."
    ^~~~~
Marlin/src/HAL/HAL_STM32F1/../../inc/SanityCheck.h:316:4: error: #error "LEVEL_CORNERS_INSET is now LEVEL_CORNERS_INSET_LFRB . Please update your Configuration.h."
   #error "LEVEL_CORNERS_INSET is now LEVEL_CORNERS_INSET_LFRB . Please update your Configuration.h."
    ^~~~~
Marlin/src/HAL/HAL_STM32F1/../../inc/SanityCheck.h:1773:6: error: #error "Enable USE_XMIN_PLUG when homing X to MIN."
     #error "Enable USE_XMIN_PLUG when homing X to MIN."
      ^~~~~
In file included from Marlin/src/HAL/HAL_STM32F1/../../inc/MarlinConfig.h:38:0,
                 from Marlin/src/HAL/HAL_STM32F1/Servo.cpp:26:
Marlin/src/HAL/HAL_STM32F1/../../inc/SanityCheck.h:45:4: error: #error "You are using an old Configuration.h file, update it before building Marlin."
   #error "You are using an old Configuration.h file, update it before building Marlin."
    ^~~~~
In file included from Marlin/src/HAL/HAL_STM32F1/../../inc/MarlinConfig.h:38:0,
                 from Marlin/src/HAL/HAL_STM32F1/Servo.cpp:26:
Marlin/src/HAL/HAL_STM32F1/../../inc/SanityCheck.h:49:4: error: #error "You are using an old Configuration_adv.h file, update it before building Marlin."
   #error "You are using an old Configuration_adv.h file, update it before building Marlin."
    ^~~~~
Marlin/src/HAL/HAL_STM32F1/../../inc/SanityCheck.h:316:4: error: #error "LEVEL_CORNERS_INSET is now LEVEL_CORNERS_INSET_LFRB . Please update your Configuration.h."
   #error "LEVEL_CORNERS_INSET is now LEVEL_CORNERS_INSET_LFRB . Please update your Configuration.h."
    ^~~~~
Marlin/src/HAL/HAL_STM32F1/../../inc/SanityCheck.h:1773:6: error: #error "Enable USE_XMIN_PLUG when homing X to MIN."
     #error "Enable USE_XMIN_PLUG when homing X to MIN."
      ^~~~~
In file included from Marlin/src/HAL/HAL_STM32F1/../../inc/MarlinConfig.h:38:0,
                 from Marlin/src/HAL/HAL_STM32F1/HAL_SPI.cpp:35:
Marlin/src/HAL/HAL_STM32F1/../../inc/SanityCheck.h:45:4: error: #error "You are using an old Configuration.h file, update it before building Marlin."
   #error "You are using an old Configuration.h file, update it before building Marlin."
    ^~~~~
In file included from Marlin/src/HAL/HAL_STM32F1/../../inc/MarlinConfig.h:38:0,
                 from Marlin/src/HAL/HAL_STM32F1/HAL_SPI.cpp:35:
Marlin/src/HAL/HAL_STM32F1/../../inc/SanityCheck.h:49:4: error: #error "You are using an old Configuration_adv.h file, update it before building Marlin."
   #error "You are using an old Configuration_adv.h file, update it before building Marlin."
    ^~~~~
Marlin/src/HAL/HAL_STM32F1/../../inc/SanityCheck.h:316:4: error: #error "LEVEL_CORNERS_INSET is now LEVEL_CORNERS_INSET_LFRB . Please update your Configuration.h."
   #error "LEVEL_CORNERS_INSET is now LEVEL_CORNERS_INSET_LFRB . Please update your Configuration.h."
    ^~~~~
Marlin/src/HAL/HAL_STM32F1/../../inc/SanityCheck.h:1773:6: error: #error "Enable USE_XMIN_PLUG when homing X to MIN."
     #error "Enable USE_XMIN_PLUG when homing X to MIN."
      ^~~~~
*** [.pio/build/STM32F103RC_bigtree/src/src/HAL/HAL_STM32F1/Servo.cpp.o] Error 1
*** [.pio/build/STM32F103RC_bigtree/src/src/HAL/HAL_STM32F1/HAL.cpp.o] Error 1
*** [.pio/build/STM32F103RC_bigtree/src/src/HAL/HAL_STM32F1/HAL_SPI.cpp.o] Error 1
`
Thanks for upgrading the configurations.

thisiskeithb, 2020-03-16:
> You are using an old Configuration.h file, update it before building Marlin.

You can't mix and match configs with different versions of Marlin. Grab the latest [Marlin 2.0.5](https://github.com/MarlinFirmware/Marlin/releases/tag/2.0.5) & [Configuration 2.0.5](https://github.com/MarlinFirmware/Configurations/archive/release-2.0.5.zip) and it will build:
<details><summary>Successful Ender-5 build log output</summary>
<pre>...
Indexing .pio/build/sanguino1284p/lib79a/libU8glib-HAL_ID1932.a
Linking .pio/build/sanguino1284p/firmware.elf
Building .pio/build/sanguino1284p/firmware.hex
Checking size .pio/build/sanguino1284p/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [===       ]  27.9% (used 4571 bytes from 16384 bytes)
Flash: [==========]  96.0% (used 124792 bytes from 130048 bytes)
================================ [SUCCESS] Took 34.72 seconds ================================

Environment               Status    Duration
------------------------  --------  ------------
mega2560                  IGNORED
mega1280                  IGNORED
rambo                     IGNORED
FYSETC_F6_13              IGNORED
FYSETC_F6_14              IGNORED
sanguino644p              IGNORED
sanguino1284p             SUCCESS   00:00:34.722
melzi                     IGNORED
melzi_optiboot            IGNORED
at90usb1286_cdc           IGNORED
at90usb1286_dfu           IGNORED
DUE                       IGNORED
DUE_USB                   IGNORED
DUE_debug                 IGNORED
LPC1768                   IGNORED
LPC1769                   IGNORED
STM32F103RC               IGNORED
STM32F103RC_fysetc        IGNORED
STM32F103RC_btt           IGNORED
STM32F103RC_btt_USB       IGNORED
STM32F103RC_btt_512K      IGNORED
STM32F103RC_btt_512K_USB  IGNORED
STM32F103RE               IGNORED
STM32F103RE_btt           IGNORED
STM32F103RE_btt_USB       IGNORED
STM32F4                   IGNORED
STM32F7                   IGNORED
ARMED                     IGNORED
STM32F103VE_GTM32         IGNORED
STM32F103VE_longer        IGNORED
mks_robin_mini            IGNORED
mks_robin_nano            IGNORED
mks_robin                 IGNORED
mks_robin_pro             IGNORED
mks_robin_lite            IGNORED
mks_robin_lite3           IGNORED
jgaurora_a5s_a1           IGNORED
STM32F103CB_malyan        IGNORED
chitu_f103                IGNORED
STM32F401VE_STEVAL        IGNORED
FLYF407ZG                 IGNORED
FYSETC_S6                 IGNORED
STM32F407VE_black         IGNORED
BIGTREE_SKR_PRO           IGNORED
BIGTREE_GTR_V1_0          IGNORED
BIGTREE_BTT002            IGNORED
teensy31                  IGNORED
teensy35                  IGNORED
esp32                     IGNORED
linux_native              IGNORED
SAMD51_grandcentral_m4    IGNORED
rumba32_f446ve            IGNORED
rumba32_mks               IGNORED
include_tree              IGNORED
================================= 1 succeeded in 00:00:34.722 =================================

Terminal will be reused by tasks, press any key to close it.</pre>
</details>

patoberli, 2020-03-16:
That was, as stated, with Marlin 2.0.4.4 and the sample configurations here. But thanks for updating the config examples for 2.0.5, will soon try those.

patoberli, 2020-03-17:
Thanks, new examples worked with 2.0.5.

closed 2020-03-17 by patoberli