MarlinFirmware/Configurations #33: ANET A8 Plus -- Bltouch non-responsive && EEPROM warnings.

RoscoeTheDog, 2020-03-14:
Hello everyone. 

I have a TriangleLabs Bltouch sensor that I am trying to install on a stock Anet A8 Plus. I am performing many upgrades at once and the motherboard has been stripped from all components besides it's USB connection to the computer while flashing the firmware. The only other component connected at this time, is the Bltouch which I have wired to the controller according to [this](http://wpcdn.cazda.com/wp-content/uploads/3D-Touch-AnetA8-Board-820x550.jpg) schematic and verified over a few other sources as well.

Marlin 2.0.4 compiles OK and uploads fine to the board using the exact configs that are attached to this thread. Upon powering the board over USB, the BLtouch sensor powers on and performs a couple of quick open and retractions, then goes into standby mode. When in standby, an Orange LED is turned on and it begins blinking a Green LED similarly to how the LED on the micro-controller does when in standby mode.

Despite the firmware uploading successfully and double, tripple, quadruple checking my resources, I cannot perform the BLtouch tests over Serial USB to verify if the hardware is working. Such tests, like echoing the state of the BLtouch sensor when using the `M119` Gcode will report the switch being 'open' when it is closed, and other Gcode that is supposed to open/close the Bltouch's "Pin" does not do anything.

I thought this could potentially be a problem with what pin is defined as the servo pin for the sensor `#define SERVO0_PIN`. Many guides for the standard A8 board claim to have this set to 27. I noticed a warning from the compiler saying that this was a redeclaration from somewhere else deeper in the config files which indicate that the pin was being set to 29 elsewhere by default. I tried both of these pin # configurations and neither one changed any behavior.

I also noticed there are some debug messages being echo'd over the serial port:

```
01:06:58.967 : echo:Unknown command: "M20"
01:06:58.970 : echo:Unknown command: "M80"
```

At one point, I plugged in the LCD display at one point and there was a warning message notifying me that there was an EEPROM version mismatch. A simple google search led me to discover an additional option in the config.h file `#define EEPROM_AUTO_INIT` which will re-initialize the EEPROM if an error is found and thus eliminate the error. However, I'm not sure if re-initializing the EEPROM would screw up the configuration set for the BLtouch. I would imagine not, since I have tried both enabling/disabling this auto-init feature anyway without further success.

What am I missing? For the Anet A8 Plus V1.7 board, is there additional setup or configuration involved? Could I be missing the wrong pin number?

I do not believe it could be a hardware or wiring issue with the BLtouch, since it does seem to successfully do appears to be a some sort of startup sequence when the micro-controller is first powered on successfully.

[marlinConfigs.zip](https://github.com/MarlinFirmware/Configurations/files/4332475/marlinConfigs.zip)

RoscoeTheDog, 2020-03-16:
It appears as though disabling the Anet LCD in the configuration file fixes the BLTouch issue. However, now I will not be able to use an LCD with my printer. Does anyone know of any information or resources about this?

RoscoeTheDog, 2020-03-16:
After some further investigation, I resolved my issue. There seems to be a lot of misinformation spreading around regarding the Anet A8 and the A8 Plus. According to _many_ **A8** owners, pin 27 on the LCD is what the signal cable (orange) should be but when looking deeper in the `pins_ANET_10.h` file, we can see this:

```
/**
 * ====================================================================
 * ===================== LCD PINOUTS ==================================
 * ====================================================================
 *
 *   Anet V1.0 controller           | ZONESTAR_LCD      | ANET_FULL_      | RepRapDiscount Full      | Thingiverse RepRap wiring
 *   physical   logical   alt       |                   | GRAPHICS_LCD    | Graphics Display Wiring  | http://www.thingiverse
 *     pin        pin     functions |                   |                 |                          | .com/thing:2103748
 *------------------------------------------------------------------------------------------------------------------------
 *   ANET-J3.1    8 ***             | N/A               | J3_TX ***       |                          |
 *   ANET-J3.2    9 ***             | N/A               | J3_RX ***       |                          |
 *   ANET-J3.3    6       MISO      | N/A               | MISO ***        | EXP2.1   MISO            | EXP2.1   MISO
 *   ANET-J3.4    +5V               | N/A               | +5V             |                          |
 *   ANET-J3.5    7       SCK       | N/A               | SCK ***         | EXP2.2   SCK             | EXP2.2   SCK
 *   ANET-J3.6    5       MOSI      | N/A               | MOSI ***        | EXP2.6   MOSI            | EXP2.6   MOSI
 *   ANET-J3.7    !RESET            | N/A               | button          | EXP2.8   panel button    | EXP2.8   panel button
 *   ANET-J3.8    GND               | N/A               | GND             | EXP2.9   GND             | EXP2.9   GND
 *   ANET-J3.9    4       Don't use | N/A               | N/C             |                          |
 *   ANET-J3.10   +3.3V             | N/A               | +3.3V ***       |                          |
 *                                  |                   |                 |                          |
 *                                  |                   |                 |                          |
 *   ANET-LCD.1   GND               | GND               | GND             | EXP1.9   GND             | EXP1.9   GND
 *   ANET-LCD.2   +5V               | +5V               | +5V             | EXP1.10  +5V             | EXP1.10  +5V
 *   ANET-LCD.3   27      A4        | N/C *             | LCD_PINS_RS     | EXP1.4   LCD_PINS_RS     | EXP2.4   SDSS or N/C *
 *   ANET-LCD.4   10                | LCD_PINS_D4       | BTN_EN2         | EXP2.3   BTN_EN2         | EXP2.3   BTN_EN2
 *   ANET-LCD.5   28      A3        | LCD_PINS_RS       | LCD_PINS_ENABLE | EXP1.3   LCD_PINS_ENABLE | EXP1.1   BEEPER_PIN
 *   ANET-LCD.6   11                | LCD_PINS_D5       | BTN_EN1         | EXP2.5   BTN_EN1         | EXP2.5   BTN_EN1
 *   ANET-LCD.7   29      A2        | LCD_PINS_ENABLE   | N/C *           | EXP2.4   SDSS or N/C *   | EXP1.3   LCD_PINS_ENABLE
 *   ANET-LCD.8   16      SCL       | LCD_PINS_D6       | BTN_ENC         | EXP1.2   BTN_ENC         | EXP1.2   BTN_ENC
 *   ANET-LCD.9   30      A1        | ADC_KEYPAD_PIN ** | LCD_PINS_D4     | EXP1.5   LCD_PINS_D4     | EXP1.4   LCD_PINS_RS
 *   ANET-LCD.10  17      SDA       | LCD_PINS_D7       | BEEPER_PIN      | EXP1.1   BEEPER_PIN      | EXP1.5   LCD_PINS_D4
 *
 *                 N/C * - if not connected to the LCD can be used for BLTouch servo input
 *                 ** - analog pin -WITHOUT a pullup
 *                 *** - only connected to something if the Bluetooth module is populated
 */
```

Notice the line in particular under the `ANET_FULL_GRAPHICS_LCD` column where it says `N/C*`. The key at the bottom states that when not connected to the LCD, this is what is to be used for the BLtouch servo. The logical pin definition is 29, not 27. By default the Anet A8 plus Marlin configuration sets the pin to 29, so nothing really needs to be changed. You just need to switch what pin you are connecting the BLtouch to.

Figuring out what the physical pin was '29' was a little trickier, but I found some photos of the pin layout [here](https://cdn.thingiverse.com/assets/27/bb/f1/ca/8a/diagram.svg). I have also uploaded the photo attached to this comment for archival purposes in case someone comes across this thread with similar issues and the photo's server goes down.
[image]

The bottom right pin-header is for the LCD. You can see it counts up to 10. You can see what logical pin it corresponds to by looking at the source code posted above `ANET-LCD.1` `ANET-LCD.2` etc etc...

In my case, `ANET-LCD.7` was what I needed to find on the LCD pin headers, which translates to `LCD-E` on the photo diagram. You can even see on the diagram itself how it's not giving accurate information on what to attach the BLtouch sensor to.

Once the BLtouch sensor is attached, you will also need to cut the corresponding wire on the LCD ribbon. If both are attached at the same time, it will not work. This is also stated in the source code: ` N/C * - if not connected to the LCD can be used for BLTouch servo input`

In conclusion, there are a **ton** of tutorials and guides spreading misinformation about this board and what pin to jump. It took me a lot of troubleshooting over a few days to figure this out. I hope it proves useful to someone else.

closed 2020-03-16 by RoscoeTheDog

thinkyhead, 2020-03-25:
That is very useful, thanks. I'll see about adding this information to our "Troubleshooting" page soon.

RoscoeTheDog, 2020-03-25:
Thingiverse is currently undergoing some web service issues. They changed the whole layout of the website and now the basic search function isn't even working. Considering COVID-19 has everyone on lockdown, it will probably take a longer time for their team of developers to resolve these problems. That is my assumption as to why the link does not work right now.

thinkyhead, 2020-03-28:
The URL works, but in most text editors it's not –ya know– "clickable."