MarlinFirmware/Configurations #35: Probing fail the moment it touches the bed. SKR 1.3 for delta printer

maviles798, 2020-03-15:
I have been trying to update my firm ware and configure marlin2.0.4 and everything works great, but im having troubles calibrating and leveling, my probe is working but the the moment the probe touches the bed. a message appears that the probing fail.

i try changing some parameters but nothing changes the problem.

Any ideas what might be the problem?

please help i accidentally deleted my last firmware 

[Configuration h and adv.zip](https://github.com/MarlinFirmware/Configurations/files/4334532/Configuration.h.and.adv.zip)

maviles798, 2020-03-16:
i found the problem

closed 2020-03-16 by maviles798

Sonsi79, 2020-03-25:
Have the same Issue.
What was your problem and how did you fix it?

maviles798, 2020-03-25:
In the pins_BTT_SKR_V1.3 the z_min_prove_pin is not define as the z_min so i just change the z_min_prove_pin to P1_25 and that fixes the problem. hope this can help you

thinkyhead, 2020-03-29:
The default `Z_MIN_PROBE_PIN` for SKR 1.3 (`P1_24`) is labeled **Z+** on the board. If you want to use the probe for homing Z you must enable the `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN` option and then you must connect the probe to pin `P1_25` labeled **Z-** on the board. Then when you use the probe, Marlin reads the `Z_MIN_PIN` (not the `Z_MIN_PROBE_PIN`).

If you disable `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN` then Marlin will read the `Z_MIN_PROBE_PIN` (**Z+** `P1_24`). If your probe is instead connected to the **Z-** plug then you need to override `Z_MIN_PROBE_PIN` in your configuration, setting it to `P1_25`.

mentaluproar, 2020-06-09:
I have the same issue but none of the above fixes have worked for me.  The probe pin is triggered, the effector stops, then lifts a bit, and it gives up, trowing a probing failed message.