MarlinFirmware/Configurations #36: inversion of home axis when endstop stuck

maudafaca, 2020-03-16:
I own a geetech gt2560 ver. 1 (with pins for bl touch) and accidentaly my Y ENDSTOP got stuck in ON position. until i descovered the problem my printer started to Y HOME on the oposite direction (Y MAX).
My endstops are the NO REPRAP type.
I think that  the least to expect was that when the endstop is activated is to be warned that the Axis MIN is activated and no movement should be visible. Especialy not to the oposite side.
Am I wrong?

thinkyhead, 2020-03-31:
G.I.G.O. The axis should basically do nothing if the endstop is stuck on. However, Marlin may be looking for a change in the state, and if no change occurs then there is no detection. It will depend on the endstop type and so on. Thanks for letting us know about the circumstance. Most testers fail to check for things like a stuck endstop or other potential bad situations.

thinkyhead, 2020-03-31:
> warned that the Axis MIN is activated

That would be perfectly normal when the axis is already homed.

maudafaca, 2020-04-02:
I understand that GRBL uses just one pin to signal AXIS:MIN and AXIS_MAX so much so, that you have to connect endstops of the NC type in serial. the software depends on machine description to know wich is the MIN or MAX endstop. I wonder if something similar might have happened here, although (and I might be wrong) I think min and max endstops use diferent pins.
As for the non changing of state, a pin high (or low) should be enough for the axis not moving in that sense or otherwise it wouldn't be a safety feature. Maybe its just GEEETECH 2560 related.
I will try to replicate the issue on other axis. Ill report back.
Thank you for your time.

tubaian, 2020-07-03:
I have a similar and maybe related issue. I have an SKR 1.4 Turbo board with TMC2209 drivers set up for sensorless homing in Marlin 2.0.x bugfix downloaded again fresh this week for a complete reinstall. The problem I have is that, after switching on my machine, the first G28 or 'home all' from my TFT35 screen causes the X and Y axes to try to home in the wrong direction and smash into the positive end of the axes. I then have to do a reset after which everything behaves perfectly. As I said, I have just done a complete reinstall with a fresh download of Marlin and a re-configuration from scratch to check that it wasn't anything I had done silly in setting up the code. I have attached my configuration.h and configuration.adv.h files. Thanks
[Marlin.zip](https://github.com/MarlinFirmware/Configurations/files/4870254/Marlin.zip)

chepo92, 2020-07-15:
> I have a similar and maybe related issue. I have an SKR 1.4 Turbo board with TMC2209 drivers set up for sensorless homing in Marlin 2.0.x bugfix downloaded again fresh this week for a complete [...]

it might be related, but you have a different setup, different board, it should be opened as a separate issue, also, you will find more support in the discord channel ;)

thinkyhead, 2020-07-15:
> I then have to do a reset after which everything behaves perfectly.

See if `M502` followed by `M500` helps.