MarlinFirmware/Configurations #41: A3S download?

tcookj66, 2020-03-24:
I purchased a used A3S and i don't want to mess it up. I bought a new power supply with a higher current rating(15 Amp) which is going to be delivered tomorrow. now I am attempting to install the software on my PC. I see that the A3S is not in the download options. Do I just download the A5S file? It has the GEN L 1.0 with the TFT28 v3.0. I am completely new at this so I don't know all of the do's and don'ts yet. i have been on the wiki and youtube

thinkyhead, 2020-03-31:
I would find an A3S user group and get the correct configuration values from them. Then take the A5S config and modify it according to that information. If you get a good, working A3S configuration together, please share!

ellensp, 2020-10-30:
For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

closed 2020-10-30 by ellensp