MarlinFirmware/Configurations #45: Impossible to compile

zonalimitatore, 2020-03-27:
Why it's impossible compile this version.
I'm using the file like downloaded,open in ide and compile.
I've got hundered of errors: 

**warning: this use of #defined may not be portable [-Wexpansion-to-defined]**

and i have to force close ide 'cause still not response....

But....You try the release before public?
I mean...If a noob starting Marlin's word from this,i think switch back early

Thanks

Sudzup1, 2020-03-28:
I am diffently a noon and I have installed it on me Ender 3 Pro stock 8bit board. I had sone issues with it but was doable.male sure your processor is the correct one. Read through the errors it tells you everything you need to know

thinkyhead, 2020-03-28:
The recommended method to compile Marlin is to use PlatformIO in Visual Studio Code along with the Auto Build Marlin extension. This is detailed on the website. If you experience problems while trying to build Marlin there are several support resources available:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if it turns out you've found a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose) to the main project.

closed 2020-03-28 by thinkyhead