MarlinFirmware/Configurations #62: Using STEALTHCHOP_XY disables x movement during z homing

emaayan, 2020-04-05:
when trying to home z, the x motor starts the move the then stops right away. 
the issue is not observed when STEALTHCHOP_XY is disabled 

hardware is skr turbo 1.4 tmc2208 FOR X,Y,Z (TMC2209) for extruder
[Marlin.zip](https://github.com/MarlinFirmware/Configurations/files/4434290/Marlin.zip)

Stieges, 2020-04-12:
I'll changed the Hybrid threshold value to 5 instead of 3 and it was gone. Probably it helps you!

emaayan, 2020-04-12:
> I'll changed the Hybrid threshold value to 5 instead of 3 and it was gone. Probably it helps you!

my question why di the sudden change, this behaviour started happening from version 2.0.4.3 and beyond

thinkyhead, 2020-04-13:
There are a lot of moving parts in a firmware and in a 3D printer, and various things can affect current, sensitivity of hybrid and stallguard, and so on. A faster move with a higher jerk or change to the junction deviation, a loss of tuned values previously stored in EEPROM… So many things can have an effect.

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose) at the main project.

closed 2020-04-13 by thinkyhead

emaayan, 2020-06-12:
i tried changing the z_threshold to 5, didn't work