MarlinFirmware/Configurations #77: Tool for understanding: feedrate, acceleration and jerk

daleckystepan, 2020-04-19:
Hi guys,

I've written basic app to better understand basic motion principles in Marlin. 
https://github.com/daleckystepan/MarlinMotion

https://github.com/daleckystepan/MarlinMotion/raw/master/screenshot.png

It is written in Python, based on simply RungeKutta ODE solver. Jerk is used only for initial velocity which is half of jerk. It is only for one axis now. We can improve it by proper jerk, s curve acceleration or so on.

Feel free to test it and give me feedback.
Thank you

closed 2020-04-19 by daleckystepan

thinkyhead, 2020-04-20:
That seems quite useful. I'll check it out soon.

daleckystepan, 2020-04-20:
https://github.com/MarlinFirmware/Marlin/issues/17618

Sorry fot that. It is more connected to Marlin so I have recreated this here.