MarlinFirmware/Configurations #80: sketch/src/module/motion.cpp:77:71: error: 'X_HOME_BUMP_MM' was not declared in this scope

jonawald, 2020-04-22:
Hi. I'm new to Marlin. My printer has been on Reperier since I built it three years ago. I want to compile the sketch but keep getting this error. I cannot figure out what setting I need to change to make it go away.

Jonathan

`
sketch/src/module/motion.cpp:77:71: error: 'X_HOME_BUMP_MM' was not declared in this scope
 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }
                                                                       ^
sketch/src/module/motion.cpp:83:1: note: in expansion of macro 'XYZ_CONSTS'
 XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);
 ^~~~~~~~~~
sketch/src/module/motion.cpp:77:71: note: suggested alternative: 'HOME_BUMP_MM'
 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }
                                                                       ^
sketch/src/module/motion.cpp:83:1: note: in expansion of macro 'XYZ_CONSTS'
 XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);
 ^~~~~~~~~~
sketch/src/module/motion.cpp:77:80: error: 'Y_HOME_BUMP_MM' was not declared in this scope
 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }
                                                                                ^
sketch/src/module/motion.cpp:83:1: note: in expansion of macro 'XYZ_CONSTS'
 XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);
 ^~~~~~~~~~
sketch/src/module/motion.cpp:77:80: note: suggested alternative: 'HOME_BUMP_MM'
 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }
                                                                                ^
sketch/src/module/motion.cpp:83:1: note: in expansion of macro 'XYZ_CONSTS'
 XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);
 ^~~~~~~~~~
sketch/src/module/motion.cpp:77:89: error: 'Z_HOME_BUMP_MM' was not declared in this scope
 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }
                                                                                         ^
sketch/src/module/motion.cpp:83:1: note: in expansion of macro 'XYZ_CONSTS'
 XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);
 ^~~~~~~~~~
sketch/src/module/motion.cpp:77:89: note: suggested alternative: 'HOME_BUMP_MM'
 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }`

thinkyhead, 2020-04-23:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

closed 2020-04-23 by thinkyhead

jonawald, 2020-04-23:
Issue is resolved. These lines were missing from my sketch for some reason. I found them in documentation and added them to resolve the issue. I was  using the kossel_XL example file as a reference to do my build. I had meant to upload my sketch, but it didn't upload for some reason. All is going except for the stuff that isn't. I won't ask about another issue in this thread.