MarlinFirmware/Configurations #82: Geeetech A20T freezing at Stepper change

Bergstern, 2020-04-26:
Hi,
I have flashed the newest Marlin versiin to my A20T (one nozzle, 3steppers).
Everything looks fine and i can print with extruder 0 but when i try to extrude with E1 or E2 i get a freezing. I use a configurated (inverted axis for X,Y,Z TMC2208) version from the examples folder fir the A20T. It seems there is a bug in the new firmware so i cant use the other steppers? 
Perhaps someone can help me.

thinkyhead, 2020-04-27:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

Legarin, 2020-07-14:
> Hi,
> I have flashed the newest Marlin versiin to my A20T (one nozzle, 3steppers).
> Everything looks fine and i can print with extruder 0 but when i try to extrude with E1 or E2 i get a freezing. I use [...]

I have the same problem

thinkyhead, 2020-07-14:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.