MarlinFirmware/Configurations #92: Creality Ender 3 (not the V2) has #define DWIN_ENDER3_V2 enabled

swilkens, 2020-05-05:
**Issue:**
Standard Ender 3 **(V1)** configuration file has the DWIN display type enabled, but the V1 of this printer doesn't have this display type - only the V2 does.

Line 2092 in Configuration.h from the `bugfix-2.0.x` branch:
`#define DWIN_ENDER3_V2`

Probably should be:
`//#define DWIN_ENDER3_V2`

It seems this is already fixed in the `import-2.0.x` branch:
https://github.com/MarlinFirmware/Configurations/blob/import-2.0.x/config/examples/Creality/Ender-3/Configuration.h#L2092

thinkyhead, 2020-05-05:
The bugfix branch is just a rebuilt version of the import branch. If something isn't in bugfix at the moment, it will appear as soon as the branch is rebuilt in sync with the import branch.

closed 2020-05-05 by thinkyhead

swilkens, 2020-05-05:
This happens rapidly then? As currently, the config in the bugfix branch is invalid for this printer.

thinkyhead, 2020-05-05:
Typically within 24-48 hours of recent updates, depending on workload and life events.

swilkens, 2020-05-06:
Apologies, I misunderstood your earlier reply. Thank you for the clarification, and thanks for all your efforts in this project.