MarlinFirmware/Configurations #93: zone leveling request

Rubenazzzo, 2020-05-06:
Hi folks.

I have been searching many places and have not found the following (if there is, forgive me)
You could implement a leveling function in marlin in the area, I explain, for those of us who have a large bed and sometimes have to print something small, when ordered to level the table, it is time (in my case the whole table and several points, due to deformation), I propose that by means of some command delimit a parallelogram, for example:
M4000 X100 (x min) Y100 (y min) K200 (x max) L (y max) P12 (points in the area to be measured), and thus go to exclusively measure that area of ​​the bed, with this I hope that the programmers from any slicing program, develop a command to get that data out and enter it at the beginning of each G-code

thinkyhead, 2020-05-07:
At this time these are the flexible options:

The [ABL Bilinear system](https://marlinfw.org/docs/gcode/G029-abl-bilinear.html) allows you to change the boundaries of the mesh that you probe with the `L`, `R`, `F`, `B`, and `H` parameters, but it will always use the number of mesh points originally configured.

The [ABL Linear system](https://marlinfw.org/docs/gcode/G029-abl-linear.html) is only good for very flat beds, but it allows you to change the boundaries of the mesh that you probe, and also the number of points with `P`, `X`, and `Y`. The final result is always a rotation matrix.

Rubenazzzo, 2020-05-07:
thanks a lot.
I have seen those options, and I never tried them, and yesterday when proposing this, I did not remember it.
sorry for the inconvenience

closed 2020-05-12 by thinkyhead