MarlinFirmware/Configurations #96: Clearly identifying which printer each configuration is for...please

gwideman, 2020-05-08:
Could there please be a convention for a comment near the top of each Configuration.h and Configuration_adv.h that prominently identifies which printer each is for? 
That comment could simply be the directory path that these files are distributed under within the "config/examples" directory.
I realize that at least Configuration.h has the string STRING_CONFIG_H_AUTHOR. However, Configuration_adv.h doesn't seem to have anything similar.

thinkyhead, 2020-05-10:
That would be great. You can submit all of them at once, or just a few at a time, according to your preference. There's no huge demand for this, so don't feel at all rushed.

thisiskeithb, 2020-05-10:
This would be a great idea. Would one or two strings be better?

```gcode
#define STRING_CONFIG_H_MACHINE_MAKE_AND_MODEL "Creality Ender-3"
```
or
```gcode
#define STRING_CONFIG_H_MACHINE_MAKE_AND_MODEL "Creality Ender-3"
```

cn-tools, 2020-05-11:
Great idea!

I personally prefer minimum two strings.

But maybe there are some printers available where 3 strings are better?

```c++
#define STRING_CONFIG_H_MACHINE_MAKE "Creality"
#define STRING_CONFIG_H_MACHINE_MODEL "Ender-3 Pro"
```
or

```c++
#define STRING_CONFIG_H_MACHINE_MAKE "Creality"
#define STRING_CONFIG_H_MACHINE_MODEL "Ender-3"
#define STRING_CONFIG_H_MACHINE_VARIANT "Pro"
```

thinkyhead, 2020-05-12:
A default machine name comes from pins, and config files are free to override this value to get the right output in the info report:
```cpp
#define CUSTOM_MACHINE_NAME "The Machine Name"
```
I don't know how useful or useless it would be to split up the machine into make, model, and variant. I wasn't planning to make the configuration tool present hierarchical select boxes, just a list with a search field.

thisiskeithb, 2020-05-12:
> A default machine name comes from pins, and config files are free to override this value to get the right output in the info report:
> 
> ```c++
> #define CUSTOM_MACHINE_NAME "The Machine Name"
> ```

This may not line up with the Make/Model since `CUSTOM_MACHINE_NAME` is usually shortened so it fits on the status screen. For example, the Wanhao Duplicator i3 Mini & variants is just "Mini".

gwideman, 2020-05-14:
For what it's worth, I've published a Python utility to automatically add a suitable distinguishing define macro to all common files in the Configurations repo, like this:

```#define STRING_CONFIG_EXAMPLES_DIR ("3DFabXYZ/Migbot")```

It just uses the salient subdirectory under 'examples' as the identifying string, and doesn't attempt to divine vendor and model, as some of the subdirs have three levels, and others only one.  This simple scheme is good enough to distinguish the files once they've been moved or copied elsewhere.

In addition, it automates the effort to maintain this define, and can be applied by the end user, if a similar feature doesn't make it into the Configurations repo.

[https://github.com/gwideman/MarlinLabelCfgs](https://github.com/gwideman/MarlinLabelCfgs)

thinkyhead, 2020-06-04:
Ran your script on the configs. I hope it helps!

closed 2020-06-04 by thinkyhead