MarlinFirmware/Configurations #100: Marlin v1.9  DIY servo probe not compile

qulet, 2020-05-10:
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
#define Z_ENDSTOP_SERVO_NR 0
#define Z_SERVO_ANGLES {150,10}
#define Z_HOMING_HEIGHT 2
#define Z_CLEARANCE_DEPLOY_PROBE 6
#define Z_CLEARANCE_BETWEEN_PROBES 6
#define Z_MIN_PROBE_REPEATABILITY_TEST
#define AUTO_BED_LEVELING_BILINEAR
#define EXTRAPOLATE_BEYOND_GRID
#define Z_SAFE_HOMING
#define HOMING_FEEDRATE_Z (6*60)
#define NUM_SERVOS 1
#define SERVO_DELAY { 1000 }
#define DEACTIVATE_SERVOS_AFTER_MOVE

#error "Z_ENDSTOP_SERVO_NR is now Z_PROBE_SERVO_NR. Please update your configuration."

qulet, 2020-05-10:
[Marlin_K_1.9_sonda.zip](https://github.com/MarlinFirmware/Configurations/files/4606745/Marlin_K_1.9_sonda.zip)

thisiskeithb, 2020-05-11:
The error is telling you exactly what you need to do: Replace `Z_ENDSTOP_SERVO_NR` with `Z_PROBE_SERVO_NR` in your config.

qulet, 2020-05-11:
thisiskeitb

what exactly does that mean
change in value
 #define Z_ENDSTOP_SERVO_NR 1
#define Z_PROBE_SERVO_NR 0
on:
 #define Z_ENDSTOP_SERVO_NR 0
#define Z_PROBE_SERVO_NR 1
gives nothing ...

qulet, 2020-05-11:
What should the configuration update consist of?

thisiskeithb, 2020-05-11:
> what exactly does that mean?

It’s been renamed.

Delete the `Z_ENDSTOP_SERVO_NR` variable and put `Z_PROBE_SERVO_NR` in its place.

qulet, 2020-05-11:
Thank you helped
 compilation was successful !!!!

qulet, 2020-05-11:
I have one more problem with Marlin 2.0 MKS Sbase V1.3 + RRDSC 12064.
  The LCD menu works, the SD card does not work. SD cable is:
P0.8 to J8-3 P0.7 to J8-2
P0.9 to J8-4 (P0.5 to J8 -4)
P0.27 to J8-5
What should I do to make the SD card work ????

closed 2020-05-12 by thinkyhead