MarlinFirmware/Configurations #98: Monoprice select mini v1 unresponsive after flashing

veeectorm, 2020-05-10:
I flashed using latest marlin and the configs from the repo.
Build goes perfectly (used the guide at https://github.com/MarlinFirmware/Configurations/tree/release-2.0.5/config/examples/Malyan/M200
It seems to flash "fine" Showing v99.109.2 after "completing" but it seems to be looping after (I can hear the steppers locking and unlocking) Temps read as 0 0, and the firmware version seems to be coming up every 5 seconds or so.

I was able to flash back the original v26 firmware.
Not sure if im missing anything. I followed those instructions to a T.

Any debug tips?

thinkyhead, 2020-05-12:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-05-12 by thinkyhead

veeectorm, 2020-05-12:
> Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If [...]

Still the same after using bugfix branch. Not sure where to go from there.

thinkyhead, 2020-06-04:
🤷  Please take advantage of the community to help you find and fix the issue. Someone probably has the same machine and might know what is wrong.