MarlinFirmware/Configurations #109: [FR] Add support for BTT SKR Mini E3 V2.0

swilkens, 2020-05-23:
Looks like SKR brought out a new version of the SKR mini E3 board for the Ender 3, now has additional fan control and a second Z-Axis connector, as well as an onboard AT24C32 eeprom.

**Manual:**
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V2.0/Hardware/BTT%20SKR%20MINI%20E3%20V2.0%20Instruction%20Manual.pdf

**PINS:**
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V2.0/Hardware/BTT%20SKR%20MINI%20E3%20V2.0-PIN.pdf

thisiskeithb, 2020-05-28:
See https://github.com/MarlinFirmware/Marlin/pull/18088 for the E3 Mini V2 pins PR (currently working out why the onboard EEPROM causes a no-boot with current Marlin. Lots of code has changed since BTT's fork from March) and V2 config: https://github.com/MarlinFirmware/Configurations/pull/111.

CTPhotography, 2020-06-25:
> See [MarlinFirmware/Marlin#18088](https://github.com/MarlinFirmware/Marlin/pull/18088) for the E3 Mini V2 pins PR (currently working out why the onboard EEPROM causes a no-boot with current Marlin. Lots of code has changed since BTT's fork from March) and V2 config: #111.

Is there any update on when a stable firmware will be available?

I'm finding numerous issues with the current on my ender 3 with the V2 
1. It doesn't always print from when i select my print in the menu, i have to turn it off then back on and then it'll print

2. For some reason it doesn't raise the hotend when moving from multiple parts of a print, leading to blobs all over my prints.

Hopefully something is in the works,
thanks a lot for the v1.2 firmware