MarlinFirmware/Configurations #110: Malyan M200 build fails without nozzle parking. Remaining config does not function as expected

JamesHagerman, 2020-05-24:
Hi there,

I'm attempting to compile the `bugfix-2.0.x` branch using Platformio. The build fails without `#define NOZZLE_PARK_FEATURE` being enabled.

I don't know enough about Marlin to know how to report this otherwise.

On that note, the latest stock `bugfix-2.0.x` branch of Marlin using the latest configs from this repo do not work on my MPSM V2 even after enabling the nozzle parking. I'm using Octoprint to control the printer and about the only thing that seems to work is the Z-axis motor. The However, the Z endstop doesn't work, the extruder fan doesn't turn on, and both the X and Y steppers just "clunk" instead of going in a particular direction. I haven't tested the extruder stepper or heaters at this point.

James

JamesHagerman, 2020-05-27:
After poking around at this more, I found what seems to be the primary issue! The following config values need to be set to false in `Configuration.h`:

```
#define DISABLE_X false
#define DISABLE_Y false
#define DISABLE_Z false
```

Once that was set, I had to flash a few times to get the stepper motor directions set correctly. My MPMS V2 ended up using:

```
#define INVERT_X_DIR true
#define INVERT_Y_DIR false
#define INVERT_Z_DIR false
```

Again, just to be clear, I also had to uncomment/enable the `NOZZLE_PARK_FEATURE` to enable it or the compile would fail:

```
#define NOZZLE_PARK_FEATURE
```

After all of that, PlatformIO (Core `4.3.4`, Home `3.2.2`, ST STM32: `6.1.0`) was able to compile a working `firmware.bin` by clicking `Build` on the `env:STM32F070RB_malyan` config! Again, I used the `bugfix-2.0.x` branch of both repos as a starting point (Marlin: `84c43e9ee799bdbaa1e5d53f73b63cca85a57d40`, Configurations: `38e200587e26d339e4960a7abad77c728e3812f9`).

Oddly, the `Fan on` and `Fan off` buttons in Octoprint didn't seem to control either fan but the hotend fan did come on when it started heating so I'm not worried.

I'm running a test print now. So far, the extruder stepper seems to be having a more difficult time (clicking) while running Marlin than it did with the stock firmware...

Test print worked! A few clicks from the extruder, but other than that, it went well!

thinkyhead, 2020-06-04:
I've had a hard time getting a working M200 build. Could you ZIP up your project and drop it on a reply so I can compare to my non-working situation?

thinkyhead, 2020-06-04:
I'm not sure about the stepper direction inverting settings, since it's so easy to have the plugs flipped backwards on one printer vs. another. We'll need to get feedback from more people in the M200 user groups so we have a larger sample size.

closed 2020-06-24 by thinkyhead