MarlinFirmware/Configurations #112: Creality CR-20. Add note to MKS_MINI_12864

Splashyn, 2020-05-24:
The CR-20 has very little comments in the config.h, so of course to get started one would use a template.
Virtually all of them use the MKS_Mini as the display of the CR-20.

I have two requests regarding this:

1.) extent the comment to include what printers typically use these

to make it easier for self-build configs

2.) add a note that Creality has their extension connectors 180° assbackwards on the stock LCD. (not the cables)

You have to cram the connectors into the sockets flipped for it to work with other Motherboards. Learned that the hard way and lost three days to tinkering with the cables and software, as there zero information ragarding the CR-20 on the internet.
Thanks for listening to my rant.

[image]
[image]
[image]

thisiskeithb, 2020-05-24:
Creality probably followed Makerbase/MKS' design (which is backwards compared to everyone else in the industry). Instead of [cutting the tab](https://www.youtube.com/watch?v=k2Ms2Cl5JFQ), you can also gently pry up the connector housings and flip them 180 degrees if you don't want to modify the cable.

Splashyn, 2020-05-24:
I prefer my caveman style, but thanks for the cleaner solution :D

regarding the MKS style imitation: maybe the comment should be extended to include something like "Check extension plug orientation and adjust the LCD if required"

thinkyhead, 2020-06-05:
A comment has been added concerning the potential for just about any graphical display, board, or LCD adapter to have reversed plugs.

closed 2020-06-05 by thinkyhead