MarlinFirmware/Configurations #113: Pid autotune cases runaway above 54 Celsius

cociweb, 2020-05-25:
Hello,
I've tried to get the correct PID constans but the autotune causes thermal runaway.
I'm able to get the constans with 54 Celsius (or lower) But I would like to target it as high as possible (nearly 60 C). 
I have an Anet A8 PLUS printer, the capability of the bed heating is 60 Celsius.
Fortunatelly, The runaway protection saves the device, And it stops the printing if it reaches 60 C.
Unfortunatelly, this happens always, If I choose 55 degree or higher for bed temperature.

So nor the printing nor the autotune works If I choose the target temperature 55-60 C deg.
Do I have a faulty termistor, or is it an issue with the fw?

Thank you for your answer.

rachael7, 2020-05-25:
The PID autotune is going to try different settings, some of which won't work and will allow overshoot on the target temperature.  5C of temp overshoot really isn't that much, so this doesn't sound out of line. Does the bed still overshoot after you apply the constants you obtained from autotuning at 54C?

oldman4U, 2020-05-27:
Hi cociweb.

My experience with PIC for the bed is, that it does not really improve it. Maybe it is turned off by default because of this. The temp of the nozzle changes much faster and much more over time, so PID is important there. The bed heats up and cools down much slower, so keeping the temp at a constant level is much easier which makes PID not necessary.

Otherwise you can check the max temp which is allowed in firmware and see if it too low for your printer.

Hope this helps

thinkyhead, 2020-06-04:
I concur.

closed 2020-06-04 by thinkyhead

oldman4U, 2020-06-04:
I would never allow myself to overrule the master himself and i hate it to see all those unused tickets opened. But the same time i do not like to leave users alone in case i see a chance to help them, so i did some tests yesterday and found Bed PID also not working on my machine, where it has been working before 100%. I did not respond earlier, because i had not enough time to verify this on a second machine, but for me it looks like Bed PID is not working properly in v2.0.5.x. ( i believe i am using 2.0.5.3 on that machine). I am willing to test more and to verify this, if it is important to do so.

Please let me know and thank you thinkyhead for all the great work

thinkyhead, 2020-06-05:
PID will tend to act up if there are noisy or weak signals, so every connection matters. If you find a consistent problem and it's not fixable with hardware tweaks, then post a bug report to the main project for someone to investigate, and provide as much info as you can gather. It's important to have the issue in the queue so someone can pick it up at some point.

oldman4U, 2020-06-05:
Hi Scott.

I will do so.

Question. Is there a way to let you know in case there are inactive
unneeded tickets, so you can close them? If so, i would actively try to
find out if they are needed and notify you so you can make a final check
and close it.I would do so for the BTT TFT, SKR Pro and SKR E3 DIP which i
use and where i can give support if possible and needed.

Georg

thinkyhead, 2020-06-05:
If you tag the original author in the comment and ask them if they want to close the issue, that usually does the trick, but you can also mention me if there are other questions about the issue.

oldman4U, 2020-06-06:
Take this ticket for example. I tried to connect the owner and got a reply from someone else: https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0/issues/45

And there are many many other unused tickets which are filling up the space, which makes it hard to find a ticket which already covers "my" question. So we have several tickets related to the same issue. I am wiling to take over to clean up the tickets related to the items i own. So if it is OK for you i will ask the owners of the unused and unneeded tickets to close them and if they do not react i will add a @thinkyhead (ticket closed due to inactivity) for a final check and closing. Is this OK for you? I believe this would avoid many unneeded tickets and would make the ticket section more useful.

thinkyhead, 2020-06-15:
The problem is that the admins use a notification system that pushes *any* activity to the top, whether issues are closed or stale, or whatever…. So in the past when we have tried having volunteers scan the issue queue to nag the author, this resulted in too many stale issues coming to the top of our notifications.

So, really the best way I have found to deal with issues is to address new and active issues at the top of the list, then dive into some older issues to find things that I can fix and close, or which are waiting around for some expertise, or which just need a canned reply.

There is a small group of people who are able to help with issues well enough that they usually get them closed or solved quickly — within a few comments. I hope we get more individuals with the right combination of knowledge and temperament, but they are one in a thousand….

oldman4U, 2020-06-15:
Hi Scott.

Thank you for your explanation. Currently i try to do exactly this with all the open BTT tickets of products i also own. Hope this helps others to find already available tickets related to a problem they have and they would like to report, easier.