MarlinFirmware/Configurations #116: Ender 3 cant keep up with junction deviation

fastdriver, 2020-05-28:
#### Issue description

Yesterday my bltouch 3.1 arrived.
I installed it and flashed Marlin-bugfix-2.0.x.
Bed leveling worked fine but i noticed blobs and stringing on curvy and detailed areas.
On closer inspection the printer seems to micro-pause and that is causing the blob.
After 8 hours of tinkering with settings like speed, resolution etc. I found the fix.
Enabling #define CLASSIC_JERK fixed the issue!

#### Additional details
Board is Creality v1.1.4
[Configuration.txt](https://github.com/MarlinFirmware/Configurations/files/4695779/Configuration.txt)

thinkyhead, 2020-06-05:
There have been some patches to JD in the last few days in `bugfix-2.0.x` so please give the latest code a try. If it is still having issues, we should look at where the bottleneck is happening.

fastdriver, 2020-06-05:
Yeap its fixed on latest bugfix!

closed 2020-06-05 by fastdriver