MarlinFirmware/Configurations #124: Creality Ender 5 Pro: default Extruder steps per mm WAAAY too low.

jimmiedave, 2020-06-04:
In the example Ender 5 Pro configuration, under default steps/unit, we find:

`#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 800, 93 }`

My Ender 5 Pro stopped printing reliably after switching to Marlin 2, printing poorly, failing to adhere to bed, always wrecking first layer, jamming nozzles, very stringy.

The fix was to calibrate the filament extrusion, and the steps/unit worked out to something like 120 steps/unit, not 93. I could get it closer with a more precise calibration, but this is about right, and made the printer operable once more.

`#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 800, 120 }`

Are there other reports of this trouble with Ender 5 Pro? Could obviously be 'wrong stepper motor supplied' or a supply-chain change.

Would've been nice to know in advance.

thinkyhead, 2020-06-05:
It could be a change. We'll have to look at what Creality3D has published….

thinkyhead, 2020-06-05:
I can't find Creality's out there yet. Other configs I checked out showed "95" for Ender 4 and Ender 5 plus. They showed "93" for CR-10 V2. No specific mention of the Pro, so it's still a mystery. I might ask around on the Ender user groups what the output of `M503` says.

jimmiedave, 2020-06-05:
This guy got 105 with an Ender3, saya all Creality machines ship in need of a calibration.

https://letsprint3d.net/how-to-calibrate-the-extruder-steps-ender-3-5-cr-10/

The Ender 5 Pro ships with an aluminum extruder (the plain Ender 5's is plastic), so it wouldn't take much of a part change to throw this off.

Another possibility: After all the trouble I mention above, I'm running a new Micro-Swiss (same .4mm size) nozzle, and I'm imagining a very minor difference in nozzle size ('cuz it's so tiny) could result in a need to recalibrate the Extruder feed.

Ideally, one would calibrate the extruder steps with no nozzle, and compensate for nozzle variation someplace else. 

Or maybe I just need to calibrate E any time I switch nozzles.

Another possibility: Perhaps Creality calibrates with PLA and now that I'm printing PETG, it's perhaps slicker and doesn't feed as well.

I'm not committed to being "right" about this - but 93 seems way low compared to the 120 steps/mm that actually delivers 100mm of filament when asked for 100mm.

Open to any thoughts, including closing. Just wanted to raise this as a possible issue.

thinkyhead, 2020-06-05:
I spoke with Luke Hatfield last night ( https://youtu.be/KsQZ46OcuNM?t=5115 ) and he had no knowledge of any normally-shipping extruder that drives at 120 steps per mm, only values in the range of ~95.

If you want to check the real mechanical value yourself, open up the extruder and measure the diameter of the filament wheel at the point where it pushes the filament. Observe the wheel as it drives the filament to see how well it complies.

The amount of filament per revolution is **π * diameter**, so if the diameter is 11mm you will push 34.557mm per revolution of the Extruder motor. With a typical 16 micro-steps x 200 steps you get 3200 ÷ (pi * 11) == 92.5992396171 steps per mm.

To justify a value of 120 steps-per-mm, the size of the filament drive wheel must be only 5mm, which is impossible, or you would need to have a 2:1 reduction (e.g., 32 micro-steps) with a 10mm drive wheel.

jimmiedave, 2020-06-05:
Good info, thank you.

I'm going to dig into this on the machine over the next couple days. If I don't have something measurable and reproducible by Mon. I'll close.

thinkyhead, 2020-06-05:
Another thing you probably already checked, but make sure your slicer knows your filament is 1.75mm and not 3mm diameter.

jimmiedave, 2020-06-06:
Slicer knows filament is 1.75m

Looking into the extruder, measuring with a micrometer, barely under 11mm (in the thousandths). Extruded filament with the bowden tube off, and 120 steps is clearly too high. Did the math to back out the excess, and got 92.59, really close to the Marlin default.

Am going to close this and look for another cause for extrusion trouble - maybe stepper current or faulty motor, or an unresolved clog (though I've shotgunned the whole hot-end/nozzle/tube/throat by now).

Thanks very much for your patient assistance with this. Sorry for pulling the alarm needlessly.

closed 2020-06-06 by jimmiedave

jatchison67, 2024-12-11:
My Ender 5 Pro keeps resetting itself back to the factory 93. Any ideas why? Or what I could do about it?

jimmiedave, 2024-12-11:
No idea except this:

If you're adjusting from the Ender screen and button, you're saving the values to EEPROM, right? 

If you're adjusting from a console (like the one built in to OctoPrint, f.e.), you're issuing an M500 after setting, to save it to the EEPROM, right? 

Failing to save to EEPROM after changing will make the machine load up whatever was in the EEPROM previously, the next time it's powered on.

DC

jatchison67, 2024-12-11:
I'm doing it from the screen and button. Is there something special to do to save it to EEPROM?

jimmiedave, 2024-12-11:
Press button to enter menus
Configuration menu
Store settings

I'm on Marlin 2.1.2.3

The menus change a little with different versions. Store settings is what you want. 

DC

jimmiedave, 2024-12-11:
[image]

jatchison67, 2024-12-11:
That's what I was missing. Thanks!