MarlinFirmware/Configurations #139: Ender 3 v2 Board Unknown

lreid7030, 2020-06-27:
Hi,

I am trying to upgrade my Ender 3 v2 to Marlin v2.0 - but the board in the config is set to Creality v4 which is not defined in boards.h.

thisiskeithb, 2020-06-27:
It’s in [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/blob/8474050adbe1bd4d363397cfedefaee996ac1e98/Marlin/src/core/boards.h#L313):

```cpp
#define BOARD_CREALITY_V4             4032  // Creality v4.x (STM32F103RE)
```

lreid7030, 2020-06-28:
> It’s in [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/blob/8474050adbe1bd4d363397cfedefaee996ac1e98/Marlin/src/core/boards.h#L313):
> 
> ```c++
> #define BOARD_CREALITY_V4             4032  // Creality v4.x (STM32F103RE)
> ```

Thank you!!!

closed 2020-06-28 by lreid7030