MarlinFirmware/Configurations #141: [REQUEST] Creality CR-10 config example for BigTreeTech SKR Mini E3 2.0

Hedda, 2020-07-01:
Request Creality CR-10 config example files for the new/updated BigTreeTech SKR Mini E3 2.0 mainboard hardware.

https://www.aliexpress.com/item/4001050145015.html

https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3

BigTreeTech is marketing the BigTreeTech SKR Mini E3 2.0 as also compatible with CR-10 (as well as with CR-10S?) but do not appear to include config files(?) and is instead pointing to this (official) upstream MarlinFirmware Configurations repo:

https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V2.0/Marlin-2.0.x-SKR-mini-E3-V2.0/config

There is already Ender 3 config examples for BigTreeTech SKR Mini E3 2.0 but there are currently no configuration example files for CR-10 or CR-10S (or config examples for any other printers in the CR-10 series):

https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/Ender-3/BigTreeTech%20SKR%20Mini%20E3%202.0

Swap-File, 2020-09-01:
I recently made a config for my Ender 5 Plus with a BigTreeTech SKR Mini E3 2.0.  It's basically the existing Ender 5 BigTreeTech SKR Mini E3 2.0 config with the existing Ender 5 Plus bed size mashed together.

Would this be useful to submit a PR for?  Or are only CR10 configs being looked for?

Gamester17, 2020-09-02:
FYI, @themodfatherinc has a BTT SKR Mini E3 V2.0 config example for CR-10 / CR-10S type printer 300x300x400 here + a binary:

https://github.com/themodfatherinc/Marlin

https://github.com/themodfatherinc/Marlin/tree/2.0.x/Marlin

He posted in https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/309 

He has one disclaimer though regrding using CR-10 which as standard do not have two Z-axis motors connected to Z1:

> I should add my printer has 2 Z motors but they are on a splitter/parallel cable so they only take one stepper motor plug for both, and therefore i have the Z current set to 800 for the Z1 motor port. If you have 2 separate Z cables you may have to recompile it or i can edit it for 2Z, not sure if it will power both connectors or not.

closed 2020-10-04 by thinkyhead

Hedda, 2020-10-05:
@Swap-File I think that it would be a good idea if either yourself or someone else made pull requests for Ender 5 Plus (and 5 Pro).

Then I think all that is missing is CR-10 S4 (CR-10 with 400 x 400 x 400mm bed size) (CR-10 with 500 x 500 x 500mm bed size) or?

Gamester17, 2020-10-05:
FYI, MarlinFirmware Configurations upstream have now committed BigTreeTech SKR Mini E3 2.0 config examples for both stock Creality CR-10 and stock Creality CR-10S

https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/CR-10/BigTreeTech%20SKR%20Mini%20E3%202.0

https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/CR-10S/BigTreeTech%20SKR%20Mini%20E3%202.0

Now all that is missing is for @bigtreetech to provide precompiled SKR Mini E3 2.0 binary firmware files for CR10 and CR10S .

Gamester17, 2020-10-11:
I know this issue is closed but FYI, @Marlor posted a little late after it was merged commenting that he think that config is wrong: 

> This config looks wrong.
> 
> Some changes are personal preferences, which should stay the same as the default config.
> like the edits to `ADVANCED_PAUSE`, `PID` or preheat presets.
> 
> Some things should not be changed because the rest of the printer hardware stays the same.
> like `STEPS_PER_UNIT`,`X/Y_CURRENT` or Acceleration/jerk
> 
> Some advanced functions which should not be in an example
> `S_CURVE_ACCELERATION` and `LIN_ADVANCE` at the same time can easily drive a beginner crazy
> 
> or dangerous, like the change of max temp from 250 to 275 on the standard ptfe lined hotend

I replied that since already merged and hence that pulls request has already been closed and thus no changes can be done there now, so the best way forward from now on would be if make an edit and submit any changes as a pull request yourselves, see:

https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/CR-10S/BigTreeTech%20SKR%20Mini%20E3%202.0

Maybe also take an extra look at CrealityV1 stock board too?

https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/CR-10S/CrealityV1

Same goes for CR-10 which is based on the same config https://github.com/MarlinFirmware/Configurations/pull/237

https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/CR-10/BigTreeTech%20SKR%20Mini%20E3%202.0

https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/CR-10/CrealityV1