MarlinFirmware/Configurations #144: Initial start up problem

tubaian, 2020-07-04:
I have an SKR 1.4 Turbo board with TMC2209 drivers set up for sensorless homing in Marlin 2.0.x bugfix downloaded again fresh this week for a complete reinstall. The problem I have is that, after switching on my machine, the first G28 or 'home all' from my TFT35 screen causes the X and Y axes to try to home in the wrong direction and smash into the positive end of the axes. I then have to do a reset after which everything behaves perfectly. As I said, I have just done a complete reinstall with a fresh download of Marlin and a re-configuration from scratch to check that it wasn't anything I had done silly in setting up the code. I have attached my configuration.h and configuration.adv.h files. Thanks
[Marlin.zip](https://github.com/MarlinFirmware/Configurations/files/4872416/Marlin.zip)

Rincewind08, 2020-07-05:
I have experienced the same issue, booting up, hitting reset, then going to restore defaults and storing the new defaults allowed me to get around the issue.  Now I've got my machines running beautifully, with amazingly accurate bed levelling.  Love this build.

Sorry if you already gave this a shot, just sharing what worked for me.
Still struggling to get my encoders to work correctly.

tubaian, 2020-07-05:
Thanks Rincewind08, the first time I built Marlin 2.0.5 bugfix I had a problem with it not saving changes to the setup from the disply but, when I rebuilt it from a new download, this problem seemed to disappear. A straight reset at the start seems to correct the boot up problem but, if I forget and do a 'home all', the axes crashing into the + endstops at speed terrifies me! Hopefully this will get corrected in a future build. I've had problems with the machine getting the Z-height wrong and digging holes in my bed mat even though the configuration seems to remember the probe offset I had set. I'm just changing from a capacitive sensor to a 3D-Touch to see if that improves things.

Rincewind08, 2020-07-05:
I have had nothing but problems with the three capacitive/inductive sensors I've used. The bl/3d touch works well. I'd like a grid of pens that would descend in an x and rotate to a +, each hitting the bed and recording when each pin was hit. It would take seconds instead of 8 minutes to get an accurate bed recording.

You might want to check your pins file for your eeprom method. Make sure it's using flash or iwc and not SD. Are you using stallguard homing or endstops?

tubaian, 2020-07-05:
Thanks. I'll check the pins file - I am using stallguard homing on X & Y.

Rincewind08, 2020-07-05:
Okay. If you're booard provides a 5 volt and ground that is separate from the endstops, use those to power your probe and only attach your sense line to your end stop plug. This solved the problem for me.

thinkyhead, 2020-07-05:
For best results getting help with configuration and troubleshooting you can also use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

closed 2020-10-30 by ellensp