MarlinFirmware/Configurations #145: Anet a8-m (Dual Extruder)

desenvolvedorindie, 2020-07-05:
There are example configuration for Anet A8-m (Dual Extruder)?
Uses KFB2.0 controller

thinkyhead, 2020-07-05:
You can put that board in a printer and then use `#define MOTHERBOARD BOARD_KFB_2`.

desenvolvedorindie, 2020-07-07:
And how about others configurations?
Extruders, sensors, motor directions and others

thinkyhead, 2020-07-08:
🤷 — That's what you would have to figure out. I don't have an Anet A8-M or a KFB 2.0.

chepo92, 2020-07-15:
> 
> 
> And how about others configurations?
> Extruders, sensors, motor directions and others
here are for issues and bugs, not support, You can get some help from the discord group, 
Your setup is very specific so you'll need to make it yourself, or pay the support for a compiled/configured fw.
you can read the guide https://marlinfw.org/docs/configuration/configuration.html
there are many online tools to configure, too.

closed 2020-07-15 by thinkyhead