MarlinFirmware/Configurations #146: [FR] Improve documentation for MICROSTEP_MODES config setting

XDA-Bam, 2020-07-07:
I recently switched to a new board with TMC2209 instead of the "good old" A4988 drivers. After a bit of tinkering, I wanted to switch to 32 microsteps for the X and Y axis. I stumbled upon the `MICROSTEP_MODES` setting, but couldn't really figure out, what it does. It's not documented, which of the six values corresponds to what driver, or which board types and drivers this applies to. I later found out that it doesn't seem to apply to boards with TMC drivers (those are set by `#define X_MICROSTEPS 32` and so on). However, the current config entry

```c++
// Microstep setting (Only functional when stepper driver microstep pins are connected to MCU.
#define MICROSTEP_MODES { 16, 16, 16, 16, 16, 16 } // [1,2,4,8,16]
```

is rather vague. Is the order of values identical to [M351 command](https://marlinfw.org/docs/gcode/M351.html)? The code in `stepper.cpp` also seems to suggest that besides the values [1,2,4,8,16], those of 32, 64 and 128 are equally supported. Are they?

I would encourage a quick update of the documentation in the config file, so users can more easily see if this setting is relevant for them and which settings are currently allowed here.

thinkyhead, 2020-07-08:
As it is described [here](https://marlinfw.org/docs/gcode/M350.html)…

> "If your board has digital micro-stepping pins (`X_MS1`, `Y_MS1`, etc.), use this command to set the micro-steps."

The only boards that have old-school digital micro-stepping are these classics:

- MINIRAMBO
- RAMBO
- SCOOVO_X9H
- MKS_BASE 1.x
- ALLIGATOR_R2
- ARCHIM1
- PRINTRBOARD_G2
- 5DPRINT

XDA-Bam, 2020-07-09:
Well, that is exactly the info which is missing from the documentation. I suggest to add that list to the `M350`-page. Then, add a one-liner to the config file that 'MICROSTEP_MODES' does the same as `M350` and link to its documentation page. That should cover all bases.

thinkyhead, 2020-07-11:
Which of those boards do you have, and how did you come by it?

XDA-Bam, 2020-07-11:
I don't have any of these boards.

cae111, 2020-07-11:
What up fam

closed 2020-07-12 by thinkyhead

XDA-Bam, 2020-07-12:
So, this isn't going to be improved because I don't own that specific hardware, or am I missing something here?

thinkyhead, 2020-07-13:
I have added comments to the config to make it clear which pins are needed, and I am adding that list of boards to the documentation for `M351`. Thus this issue can be closed.

thillRobot, 2022-05-26:
I have RAMBOs on some old MakerGears and I am curious about this setting. What is the order of the of the six values in `#define MICROSTEP_MODES { 16, 16, 16, 16, 16, 16 } // [1,2,4,8,16]`? @thinkyhead do you mean the order is the same as in [here](https://marlinfw.org/docs/gcode/M350.html)?

kpdelucia, 2022-10-28:
does this mean you should not change to 32 or above if you have specific  drivers such as 2208, 2209?  I changed from 16 to 32 and I doubled my steps per mm and the parts x and y now travel half as far as they used to.  This practice was shown in a video I watched but it didn't work like its supposed to for me.   I am using BTT skr pro 1.2 and 2209