MarlinFirmware/Configurations #153: Unable to create a PR against bugfix-2.0.x branch

hub-jba, 2020-07-15:
Hi I have created some configurations to support an extensible ui port of the AI3M code base (PR on marlinfw here: https://github.com/MarlinFirmware/Marlin/pull/18655). The corresponding configurations to support this build are sitting in my repo here: https://github.com/hub-jba/Configurations

I'm trying to generate a PR for my added configs against the Configurations repo, but I don't think I am able to do so.

chepo92, 2020-07-15:
Be specific, what stops you? are you targeting the default branch (import)

hub-jba, 2020-07-15:
I was trying to target the bugfix-2.0.x branch but I have been told to target the default branch (import) instead on discord. Will do so now

closed 2020-07-15 by hub-jba

chepo92, 2020-07-15:
Yeah the instructions are quite hidden 
https://github.com/MarlinFirmware/Configurations/blob/import-2.0.x/.github/contributing.md

hub-jba, 2020-07-15:
Not sure if this is being serious or said to be funny, but for clarity's sake, I was simply trying to follow the convention set in the main marlin repository for which the far more accessible instructions on the website were very clear that you should be making PR's against the bugfix-2.0 branch. On top of that the import-2.0.x branch is rather confusingly named. Makes it look like someone set that branch up for the initial import and then forgot to change the default on the repo, which is what I had assumed happened.

chepo92, 2020-07-15:
You didn't do the search, hahahah, I had the same thinking some months ago, https://github.com/MarlinFirmware/Configurations/issues/81
I agree it is a bit confusing, I dont't know what happened, as always, ideas and PR's are welcomed

hub-jba, 2020-07-15:
Lol, that is funny to see that you had the exact same experience as me.