MarlinFirmware/Configurations #155: SKR 1.4 turbo (w/ tmc2209) isn't reading the configadv stepper current settings.

RunLoganRun, 2020-07-15:
They're always set to 800 after flashing, no matter what I set them to in the configadv TMC section. M906 changes them with no issues. 

I'm using the current bugfix version.

Also, in the main version, there's an issue compiling. It gets stuck asking for neopixel settings, which I don't use either way.

thinkyhead, 2020-07-15:
Does `M502` followed by `M500` help?

RunLoganRun, 2020-07-16:
> Does `M502` followed by `M500` help?

No, that's not the issue. I can set them fine with gcode, it's not reading the firmware presets.

thinkyhead, 2020-07-17:
Trinamic steppers can only be configured when there's full power to the stepper drivers. Is it possible that your 12V power is lagging behind the initial power to the board?

ellensp, 2020-10-30:
This Is not really the place for support questions. Please use some of the following resources. 

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

closed 2020-10-30 by ellensp