MarlinFirmware/Configurations #163: Where did the BigTreeTech Examples go?

ciaranj, 2020-07-22:
Hi! Really sorry if I'm going crazy, but I *thought* I was tracking the bugfix-2.0 branch for the base configurations of the BigTreeTech/SKR Mini E3 1.2 

PlatformIO just updated underneath me so I wanted to recompile my firmware and my usual process is to grab the example config from this repo and then apply my customisations as commits atop it... but the base examples seem to be missing? 

Is bugfix-2.0 still the correct branch to look at (I'm tracking bugfix-2.0 on the main Marlin repository.) ?

thinkyhead, 2020-07-23:
```
config/examples/Creality/Ender-3/BigTreeTech SKR Mini E3 1.2/
```

closed 2020-07-23 by thinkyhead

ciaranj, 2020-07-24:
Ah thank you, I thought I was losing my mind (and I swear I looked in the Creality folder too!) Sorry.