MarlinFirmware/Configurations #171: Compile error on the marlin 2.0.6 after added lcd

SG-Viper2020, 2020-07-30:
I have downloaded the marvin 2.0.6 version and when I add the lcd, the compilation failed to complete.
Arduino IDE used is version 1.8.10 and 1.8.13, both having same error of : 
fork/exec C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc.exe: The filename or extension is too long.
Error compiling for board Arduino Mega or Mega 2560.

The way to iterate this is by undefined lcd 

// RepRapDiscount FULL GRAPHIC Smart Controller
// https://reprap.org/wiki/RepRapDiscount_Full_Graphic_Smart_Controller
//
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

thinkyhead, 2020-07-30:
> The filename or extension is too long.

You will need to move your project closer to the root of the hard drive.

It may also help to use PlatformIO instead of Arduino IDE.

closed 2020-07-30 by thinkyhead

SG-Viper2020, 2020-07-30:
Is not solving if the project move closer to the root.
The issue comes after added LCD...
Adding other peripheral is not showing this issue.

thinkyhead, 2020-07-30:
> "The filename or extension is too long"

This is a problem with the build tool. Try PlatformIO instead.