MarlinFirmware/Configurations #183: Ender 3 v1 with board v2 STM32F103RET6

badouuuuuu, 2020-08-09:
Hello,

I have an Ender 3 v1, and when I ordered my 1.1.5, they sent me a 4.2.7 (TMC 2225) from an Ender 3 v2.

It works great with a firmware found on a [**this**](https://www.reddit.com/r/ender3/comments/i3dqnx/stepper_motors_engaging_but_not_moving_after/g0t7j8g?utm_source=share&utm_medium=web2x) reddit topic

[Board here](https://imgur.com/a/QXxLFmn)

But when I try to update the board with Marlin 2.0.6 bugfix, the screen of my Ender 3 v1 does not boot.
I tried a multiple time to make a new bin with vscode, but still not working...

Do you have an idea ?

badouuuuuu, 2020-08-09:
it's ok with another github fork bugfix

GijsvanDulmen, 2020-08-12:
@badouuuuuu Could you point towards the specific github fork bugfix? I received the same 4.2.7 board and want to update towards another Marlin build.

badouuuuuu, 2020-08-13:
> @badouuuuuu Could you point towards the specific github fork bugfix? I received the same 4.2.7 board and want to update towards another Marlin build.

I closed too quickly, no release works with the 4.2.7 board.

When I update on marlin 2.0.6 I have the error "homing failed", I tried all the config files (ender 3 v2 / Pro 1.5)

I have to reopen a ticket, I would also like to update my board and creality does not want to help me, they refer me to the amazon reseller ....

I'm working on this weird firmware : https://mega.nz/file/rNdh2ADZ#DTeOwwJ9YI7EPeQVcyvRzh_EUBuP3T8qa7RcXB8yxqc 
until marlin is fixed

badouuuuuu, 2020-08-13:
> @badouuuuuu Could you point towards the specific github fork bugfix? I received the same 4.2.7 board and want to update towards another Marlin build.

Can you manifest your issue here : https://github.com/MarlinFirmware/Configurations/issues/185

closed 2020-08-13 by badouuuuuu