MarlinFirmware/Configurations #185: Ender 3 v1 with 4.2.7 board

badouuuuuu, 2020-08-13:
I received a creality 4.2.7 motherboard instead of a 1.1.5, but I cannot update marlin 2.0.6.
When I use the Ender 3 Pro 1.5 / Ender 3 v2 configs files, the axes do not work.
And it is impossible to make a home, I have the error "_Homing failed_" ...
I tried several times to build new firmware, but still not working...

Board Creality v4.2.7

[image]

[image]

I have no problem with this firmware: https://mega.nz/file/rNdh2ADZ#DTeOwwJ9YI7EPeQVcyvRzh_EUBuP3T8qa7RcXB8yxqc

[image]

GijsvanDulmen, 2020-08-13:
I also received the 4.2.7 after ordering a 1.1.5 silent motherboard. The firmware which was on the device when it got delivered had the same "homing failed" error. I did hear a sound when it started but nu moving stepper motors. I installed the same firmware as mentioned by @badouuuuuu above and got it to work (showed 0.0.6 version marlin as well).

I'm not flashing it with marlin just yet until I know what configuration I exactly need :-)

badouuuuuu, 2020-08-13:
Go to for the issue : https://github.com/MarlinFirmware/Marlin/issues/19015