MarlinFirmware/Configurations #189: [REQUEST] Tevo Tarantula Pro BLTouch Configuration.h

DouglasDRF, 2020-08-18:
Greetings. I would like some help in Configuration.h file params for using a BLTouch probe like. It's not the one bundled by Tevo, it's a generic one that I assembled.

The printer come with a sample of Configuration.h in SDCard but for Marlin 1.x not the 2.x and I'm affraid of having breaking changes.

Did somebody already set up BLTouch in Tevo Tarantula Pro in Marlin 2.x?

DouglasDRF, 2020-08-18:
I've tried to use the configurations of factory firmware but I got something going wrong. The sensor seems to work properly, On turn on, it executes the self-testing pulling up and pullng down the probe twice. In the Marlin's interface, in configurations menu, I can Deploy or Stow the probe sucessfully. However. When I go to Motion and Auto Home (to all axis) or home Z axis only, it stow the probe fully, glowing red and beeing upside the nozze, but the right behaviour should be full deployed downside the nozzle with the light turned off.

DouglasDRF, 2020-08-18:
I checked 
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
And it was commented, so I enabled it. Now the probe is being deployed but when I home when the probe touches the surfacec it's not stopping the Z-Axis

DouglasDRF, 2020-08-19:
I figured out the the white and black pin which sends the signal where inverted, not in the board, but in the GX12 socket by default in printer. I inverted on the junction the black and white wires and it worked.

closed 2020-08-19 by DouglasDRF

JAbelP, 2022-09-17:
@DouglasDRF Could you send me your file? My config isn't working super well anymore