MarlinFirmware/Configurations #194: Ender 3 Pro build fails with Power Loss Recovery

pmjdebruijn, 2020-08-28:
In file included from Marlin/src/HAL/STM32F1/../../inc/MarlinConfig.h:41:0,
                 from Marlin/src/HAL/STM32F1/HAL_SPI.cpp:35:
Marlin/src/HAL/STM32F1/../../inc/SanityCheck.h:2183:6: error: #error "POWER_LOSS_RECOVERY is incompatible with SDCARD_READONLY."
     #error "POWER_LOSS_RECOVERY is incompatible with SDCARD_READONLY."

The issue occurs here, as POWER_LOSS_RECOVERY is defined after it is checked, thus NONE erroneously concludes it's okay to set SDCARD_READONLY (mind the line numbers):

https://github.com/MarlinFirmware/Configurations/blob/import-2.0.x/config/examples/Creality/Ender-3%20Pro/Configuration_adv.h#L1136

```
  #if NONE(POWER_LOSS_RECOVERY, BINARY_FILE_TRANSFER, SDCARD_EEPROM_EMULATION)
    #define SDCARD_READONLY                 // Read-only SD card (to save over 2K of flash)
  #endif

```

With this decommented (modified):
https://github.com/MarlinFirmware/Configurations/blob/import-2.0.x/config/examples/Creality/Ender-3%20Pro/Configuration_adv.h#L1167

```
  //#define POWER_LOSS_RECOVERY
```

maweil, 2020-08-30:
Thanks for analyzing this. The check in line 1136-1138 [https://github.com/MarlinFirmware/Configurations/blob/863313fca04e35430a78668cae1d7d6bcc52a1d1/config/examples/Creality/Ender-3%20Pro/Configuration_adv.h#L1136-L1138](https://github.com/MarlinFirmware/Configurations/blob/863313fca04e35430a78668cae1d7d6bcc52a1d1/config/examples/Creality/Ender-3%20Pro/Configuration_adv.h#L1136-L1138)

 would need to be placed somewhere after line 1298: [https://github.com/MarlinFirmware/Configurations/blob/863313fca04e35430a78668cae1d7d6bcc52a1d1/config/examples/Creality/Ender-3%20Pro/Configuration_adv.h#L1298-L1311](https://github.com/MarlinFirmware/Configurations/blob/863313fca04e35430a78668cae1d7d6bcc52a1d1/config/examples/Creality/Ender-3%20Pro/Configuration_adv.h#L1298-L1311)

With this change, the firmware builds with `POWER_LOSS_RECOVERY` or `BINARY_FILE_TRANSFER` enabled (using `MELZI_OPTIMIZED`).
 
For the default `MELZI_CREALITY` motherboard of the Ender 3 and Ender 3-Pro `SDCARD_EEPROM_EMULATION` is not needed and therefore always undefined.

thinkyhead, 2020-09-10:
Because not all these conditions can be known within `Configuration_adv.h` users will just have to set the option themselves if they determine that they can.

closed 2020-09-10 by thinkyhead