MarlinFirmware/Configurations #195: Use Controller Fan error

Dogsrule333, 2020-08-28:
Have an odd issue when compiling for Ender 5.  I get a SanityCheck.h error 
#error "USE_CONTROLLER_FAN requires a CONTROLLER_FAN_PIN. Define in Configuration_adv.h."

Not sure what to make of this.

Only things I adjusted were: Changing the board from from the skr mini v2.0 to the v1.2, changed the bed size to 235, And then changed the envs to STM32F103RC_btt in platformio.ini

Any thoughts.  Thanks.

thisiskeithb, 2020-09-04:
The SKR Mini E3 1.2 board & below do not have a `CONTROLLER_FAN_PIN`. This is a feature of the V2 board only, so you'll need to comment out `USE_CONTROLLER_FAN`.

Dogsrule333, 2020-09-04:
Many Thanks.  I thought that was what I might have to do.  But didn't want it to mess up the system, until I knew for sure.

Went into config adv.h and found Use controller fan, and disabled it.  Will try it tomorrow.  Thanks again.

thisiskeithb, 2020-09-04:
There are other changes between the V2 and 1.2 board configs like `*_SLAVE_ADDRESS` for each axis as well as some others, so you'll probably have to make some more changes.

Take a look at the SKR Mini E3 V1.2 & V2.0 configs in the Ender-3 config folder to get a better idea of what else needs to be changed: https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/Ender-3

Dogsrule333, 2020-09-04:
Ok.  Thanks.  The Use Controller Fan was the only error that flagged during the firmware build.

thisiskeithb, 2020-09-04:
The other settings won't cause errors, but will mess with your printer's ability to print correctly. If you were using the V2 board & config and switched to a 1.2 board, I'd recommend starting over the 1.2 config.

Dogsrule333, 2020-09-04:
Ok.  So, I had grabbed the configuration files for the Ender 5, which only have a file for the 2.0 board.  But within the data is the info for the 1.2.  I changed both the board and platformio.ini to match for the 1.2.  Are you saying that there still are issues by doing that?

thisiskeithb, 2020-09-04:
Yes. They are different boards that use different settings for TMC drivers & such since one uses hardware serial and the other doesn't. You can't just change the `MOTHERBOARD` setting & PIO environment.

Dogsrule333, 2020-09-04:
Ok.  Any thoughts on why there isnt a folder for the 1.2 under the Ender 5 File?

thisiskeithb, 2020-09-04:
No one has created a config for that hardware combo yet.

Dogsrule333, 2020-09-05:
Fair Enough. 
Although I am surprised, since configs for Ender 5 with mini skr 1.2 have been available up to 2.0.6.  But I guess that is the joys of programming.  I will just keep using 2.0.5 until someone smarter then me creates the config.  Thanks for all the replies.

closed 2020-09-10 by thinkyhead