MarlinFirmware/Configurations #197: Anet A9 support??

darknessblade, 2020-08-30:
Hello, i noticed there was no entry for the Anet A9 on the [Configurations/config/examples/Anet/] tab.
From what i know and read online it looks like it uses the same board/hardware as the A8.

The printer is not on the Anet fandom wiki.  {not sure why, but it could be that it was not sold that much}
but you can find Manuals and a few reviews online.
https://www.manualslib.com/manual/1436216/Anet-A9.html

The printer is somewhere between a Tronxy X1 and a Anet A8. {in terms of bed size and hardware}
{The bed size is 160x160mm} 

___________________________
I was wondering how i can set it up correctly and if someone is willing to help me set it up Correctly. {i have minor experience with coding}

Since I recently ordered one for a good discounted price {less than 80$} {still have to receive it.

I could try to get Anet to give me the source code for the A9, but the chances of this are slim.
That way setting it up with a newer marlin would be easier.

thinkyhead, 2020-10-06:
See if the A8 config is a good starting point, but be sure to get tips from other users on Facebook or wherever there's A9 support. And if you end up with a good set of configurations please submit them here!

darknessblade, 2020-10-06:
@thinkyhead 

I did found this fork:
https://github.com/technoo10201/Anet_A9_Marlin_Firmware

But when i try to Compile it in Visual studio code i get a load of errors.

Did a test compile of the A8 version, and it gave no errors.

So  i tried manually recreating the A9 firmware, with the A8 as a base.
but this gave the same amount of errors.

thinkyhead, 2020-10-23:
I found the errors in that branch pretty easy to resolve, but the build was too large to fit on a Melzi until I turned on `SLIM_LCD_MENUS`. Here are the modified configs:

[Configurations.tar.gz](https://github.com/MarlinFirmware/Configurations/files/5426986/Configurations.tar.gz)

closed 2020-10-23 by thinkyhead