MarlinFirmware/Configurations #200: V1.5 for regular Ender 3

ameyp, 2020-09-04:
The Ender 3 Pro's are not the only ones that have been shipping with the new 4.2.2 boards. I received a regular Ender 3 with the new board, and apparently all Ender 3's and 3 Pro's going forward will have this board https://www.creality.com/blog-detail/ender-3-motherboard-change-notification

thisiskeithb, 2020-09-04:
To add to this, if you buy the silent board by itself, it will be a 4.2.7 board with TMC2225s instead of 2208s and a couple other changes.

Creating configs with all these variations will be a chore.

ellensp, 2020-10-05:
v4.2.7 variant is now here https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/Ender-3/CrealityV427
I suspect all that is needed this change this to a V4.2.2 would be to change   
`#define MOTHERBOARD BOARD_CREALITY_V427 `
to   
`#define MOTHERBOARD BOARD_CREALITY_V4`
and perhaps some axis inversion.

@ameyp could you give this a try?

yysh12, 2020-10-07:
Don't even do that. Just use the Ender 3 Pro V1.5 config files. The Ender 3 Pro and Ender 3 are exactly the same with the exception of a few minor cosmetic changes in construction, but otherwise will work the same.

yysh12, 2020-10-07:
So if anything simply relabel to Ender 3/Ender 3 Pro V1.5

yysh12, 2020-10-20:
Speaking of which, it appears a lot of the Ender configs can actually be consolidated down into a single config. It would probably make it easier for the devs to maintain. I noticed substantial differences between the Ender 3 and Ender 3 Pro/V1.5 configs when in reality all that changed between the boards are the drivers. For example, some of them default to classic jerk and others default to junction deviation. From my understanding the Ender 3 Series are practically the same across the range with minor cosmetic and driver differences.

thinkyhead, 2020-10-23:
I would not be entirely averse to configurations using the `MB()` macro to modify settings that need to be modified for different boards, and then having all current machine configs that differ only by `MOTHERBOARD` squashed down to single configs. It depends on how many changes there really are and how messy it gets.

closed 2020-10-23 by thinkyhead